verilator/verilator #2673: Multiple assigns to same packed struct misordered

yanx21, 2020-12-06:
Hi:

This is a issue very similar to #2576 

Testcase:
```verilog
module t (
    input          clk,
    input   logic  ready,
    output  logic  rOut
);

struct packed {
    reg rIn;
    reg rOut;
    reg ready;
} some;

reg rIn;
always @ (posedge clk) begin
    rIn <= ~rIn;
end

always @ (*) begin
    some.rIn = rIn;
    rOut     = some.rOut;
end

always @ (*) begin
    some.ready = ready;
    if (some.ready == 1)
        some.rOut = some.rIn;
end

endmodule
```
The input signal `ready` is equal to 1 during the evaluation, and here is the waveform:
[image]
As you can see, value of packed struct `some`, changes both at the posedge of clk and negedge of clk, which must be wrong.

I think the reason is the same with #2576. In that issue, your workaround is adding commnet `/*verilator_split_var*/`, which is useful to that case, but useless to this one.

I know it's a long term work to fix this bug, but can you add some mechanisms to get this around. For example, make `/*verilator_split_var*/` work when using packed struct.

wsnyder, 2020-12-06:
Agreed, the scheduler stuff is already well bugged.

>make /*verilator_split_var*/ work when using packed struct.

That's a good point, let's use this bug for that, @yTakatsukasa perhaps you have a chance in the future to look at this? Thanks.

yTakatsukasa, 2020-12-09:
Hmm, I used the latest master ff3d35ca615888615603d8fdb5800a61091e9de0 and split_var looks working.

## The entire code
To avoid misunderstanding, I paste the entire code including testbench. The design is as same as https://github.com/verilator/verilator/issues/2673#issue-757865099. Just tracing and main() is added.
```systemverilog
module t (
    input          clk,
    input   logic  ready,
    output  logic  rOut
);

struct packed {
    reg rIn;
    reg rOut;
    reg ready;
} some 
`ifdef USE_SPLIT_VAR
/*verilator split_var*/
`endif
;

reg rIn;
always @ (posedge clk) begin
    rIn <= ~rIn;
end

always @ (*) begin
    some.rIn = rIn;
    rOut     = some.rOut;
end

always @ (*) begin
    some.ready = ready;
    if (some.ready == 1)
        some.rOut = some.rIn;
end

int count = 0;
always @(posedge clk) begin
    count <= count + 1;
    if (count == 10) $finish;
end

initial begin
    $dumpfile("vlt_dump.vcd");    
    $dumpvars();
end

`systemc_implementation
#include <memory>
#include "Vt.h"
uint64_t tick = 0;
double sc_time_stamp() { return tick; }
int main(int argc, char *argv[]) {
    auto top = std::make_unique<Vt>("top");
    Verilated::traceEverOn(true);    
    Verilated::commandArgs(argc, argv);   
    top->ready = 1;
    while(!Verilated::gotFinish()) {
        ++tick;
        top->clk ^= 1;
        top->eval();
    }
    return 0;
}
`verilog
endmodule
```

## run without split_var
Verilate with the following command,
> verilator --trace -Wno-fatal --cc --exe --top-module t src/t.v --build

I get the same waveform as https://github.com/verilator/verilator/issues/2673#issue-757865099 .
[image]

## run with split_var
Verilate with the following command that defines USE_SPLIT_VAR,
> verilator --trace -Wno-fatal --cc --exe --top-module t src/t.v +define+USE_SPLIT_VAR --build

[image]

`some` is updated only at the rising edge of `clk`.

I'm not yet sure what is the missing part to let split_var work on your side.

yanx21, 2020-12-28:
OK. Appreciate your reply.
I tried your code, and split_var does work.

closed 2020-12-28 by yanx21