verilator/verilator #2676: Verilator as fault simulator

endrikaja1, 2020-12-07:
First of all, let me thank everybody contributing to this fast hardware simulator. Me and my colleagues use many of Verilator features and that's why we would like to also contribute to the community.

**Background**
We have been using Verilator as a fast fault simulator for some time now. Main idea behind our approach was to modify the source code such that it turn all signals into objects of a class _fi_object_ , e.g:
   instead of: IData/*31:0*/ top__DOT__sub__DOT____Vtogcov__count_c
   to:             fi_object<IData> top__DOT__sub__DOT____Vtogcov__count_c;

By using assignment operator overloading, we were able to model different kind of faults (e.g. stuck-at-0, stuck-at-1, bitflips, intermittent_faults), so basically object oriented saboteurs. To inject the faults, simply some values should be assigned to this class members, such as fault location, injection time.

An execution overhead of 40% was added while simulating a complete SoC and went down to about 12% on simpler designs.

We would like to ask about any opinion regarding the approach and would be happy to share the solution.

Cheers!

wsnyder, 2020-12-07:
That's a cool technique.  Yes, please share, is there a paper or writeup on the larger approach?  (Also if you don't mind please set your github name so we know you?)

These things come to thought,
1. #2660 might be useful as you could do multiple runs at once or compare state between them (if state settles no reason to keep simulating).
2. Are you only annotating storage cells as opposed to combo signals?  For soft error purposes combo signals don't matter, and removing them might speed it up a lot.
3. If you can make a reasonably clean pull set to add a custom wrapper object ("fl_object") we can consider supporting it.

endrikaja1, 2020-12-08:
Thanks for replying. So to answer your questions:

Actually for the moment there is no paper, nor documentation, but it is not a lot of effort to do that. Name is set :) .

1. Yes it could be useful. Also for each fault injected (1 fault per simulation), there is no need to generate new hw model, so we save some time also by not generating .cpp and .h files.

2. Actually, all signals that are declared in the main header file are turned into objects of the class, since we wanted to make it generic (we do not know what fault injection campaign the user intends to do). 

- Basically only the top header file is changed when the user adds **-fi** argument to the executable. Source code is modified to support this argument.

- Also the new class is defined inside the header file, some very minor changes to .cpp files.

- In the testbench, the user has to simply assign values to the class members, e.g.
```
 fi_object::fault_location="verilator's generated signal name",
 fi_object::injection_time= 30 (corresponds to VL_TIME_Q),
 fi_object::fault_model="sa0", etc.
```

3. Sure we can do that. It might take some time to do some code cleaning and adding comments regarding LoC which are related to fault injection. Maybe we can share the only the class for the moment.

I hope it became more clear.

Cheers!

bogdantabacaru, 2020-12-08:
> That's a cool technique. Yes, please share, is there a paper or writeup on the larger approach? (Also if you don't mind please set your github name so we know you?)

Maybe I can help with the concept. Please have a look at Chapter 4.5.2: https://mediatum.ub.tum.de/doc/1471529/1471529.pdf

On a different but related note, Chapter 5.6 introduces a checkpoint-restore mechanism for Verilator.

keerthidevraj, 2020-12-08:
We are also writing up a paper to explain the approach with useful results. We will make it available as soon as we finish writing.

JoGei, 2020-12-10:
Hello and sorry for the long post.

It's great to see that fault injection (FI) is getting some interest. Maybe I can provide a few insights, too.
At TUM EDA, we did something similar to the operator overloaded objects replacing the simple Verilator data types (IData, CData, WData, ...). Back then, we also added some tracing functionality to the signals which were used to observe a core's state for switching in multi-level simulations (ISS<->RTL). Unfortunately, after also adding these trace capabilities, we were experiencing poor sim. performance (comparing our performance with @endrikaja1's, I'd say this was mostly caused by the complex data structures). This is the related paper: https://ieeexplore.ieee.org/document/8342081

I don't want to say the "complex data structs" or whatever we should call it is bad, but I want to throw my hat into the discussion and sketch out another approach to FI with Verilator:

After our OO, we looked into doing the injections through source code transformation on the verilated model with LLVM, which worked out quite well. Our case study back then was on Pulp's RI5C-Y RISC-V core and resulted in an overhead of ~10% compared to pure Verilated model (single threaded, O3). We found that the C++ compiler is handling source transformation much better than our first attempt with more complex data structures. The related work was published as a master thesis a while ago. Open access is here: https://mediatum.ub.tum.de/1553527

One very important issue that has to be mentioned is that the fault injection should happen asynchronously, e.g. on assignment of the variable. Reason for this is that we observed cases where a variable is read after its fresh assignment in one evaluation cycle. This happens mostly on variables near module outputs. Here is a basic example:

```
module fiapp
	(
		input logic clk, reset,
		input logic a, enable,
		output logic o1, o2, o3
	);
	logic q1, q2, q3;

	assign o1 = q1;
	assign o2 = q2;
	assign o3 = q3;

	always_ff @(posedge clk, posedge reset)
		if(reset) begin
			q1 <= 1'b0;
			q2 <= 1'b0;
			q3 <= 1'b0;
		end
		else begin
			if (enable) begin
				q1 <= a;
			end
			q2 <= q1;
			q3 <= !q1;
    end
endmodule
```
which translates with Verilator v4.102 to: (only sequent body considered)

```
VL_INLINE_OPT void Vfiapp::_sequent__TOP__1(Vfiapp__Syms* __restrict vlSymsp) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vfiapp::_sequent__TOP__1\n"); );
    Vfiapp* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Body
    if (vlTOPp->reset) {
        vlTOPp->fiapp__DOT__q2 = 0U;
        vlTOPp->fiapp__DOT__q3 = 0U;
    } else {
        vlTOPp->fiapp__DOT__q2 = vlTOPp->fiapp__DOT__q1;
        vlTOPp->fiapp__DOT__q3 = (1U & (~ (IData)(vlTOPp->fiapp__DOT__q1)));
    }
    vlTOPp->o2 = vlTOPp->fiapp__DOT__q2;
    vlTOPp->o3 = vlTOPp->fiapp__DOT__q3;
    if (vlTOPp->reset) {
        vlTOPp->fiapp__DOT__q1 = 0U;
    } else {
        if (vlTOPp->enable) {
            vlTOPp->fiapp__DOT__q1 = vlTOPp->a;
        }
    }
    vlTOPp->o1 = vlTOPp->fiapp__DOT__q1;
}
```

Consider Fault Injections (FIs) in "q1" aka vlTOPp->fiapp__DOT__q1 or "q2" or "q3" between calls to the main eval inject. Their respective outputs "o1, o2, o3" would not see the injected fault, as these are overwritten by an assignment. In contrast, a FI in "q1" could propagate correctly through "q2" and "q3" to "o2" and "o3" respectively. 
This problem can be solved with asynchonous injection "on assignment". From what I understand @endrikaja1's solution is already doing this. However, there is another issue. In the example, "q2" is only updated (ignoring resets), if the enable input is high, thus, an FI "on assignment" would not happen if the enable is low although "q1" is driving other nets. From my understanding, a soft error would not care for a register's enable input when flipping its state. Because of this, we did both: If an FI "on assignment" was not triggered, we "intermittenly" inject after the evaluation concluded.

Here is the example after running the source transformation on it:
[... 20 lines cut ...]
    if (vlTOPp->reset) {
        vlTOPp->fiapp__DOT__q1 = 0U; SEQ_TARGET_INJECT(gTD.e_fiapp__DOT__q1);
    } else {
        if (vlTOPp->enable) {
            vlTOPp->fiapp__DOT__q1 = vlTOPp->a; SEQ_TARGET_INJECT(gTD.e_fiapp__DOT__q1);
        }
    }
    vlTOPp->o1 = vlTOPp->fiapp__DOT__q1; SEQ_TARGET_INJECT(gTD.e_o1);

 	/* Intermitten (non-dominant target injections) */
	INT_TARGET_INJECT(gTD.e_o2);
	INT_TARGET_INJECT(gTD.e_o3);
	INT_TARGET_INJECT(gTD.e_o1);
	INT_TARGET_INJECT(gTD.e_fiapp__DOT__q2);
	INT_TARGET_INJECT(gTD.e_fiapp__DOT__q3);
	INT_TARGET_INJECT(gTD.e_fiapp__DOT__q1);
}
```

As the code snipped depicts, we only add to the Verilator source code and do not change the variable types. `gTD` is a global target dictionary that gets generated during the source transformation and - together with some API files - builds the whole FI schedule system. First, on each call to `SEQ_TARGET_INJECT` (which is the "on assignment" FI), a check is performed whether an injection has to be performed. Secondly, after the sequent body all related variables are again checked for correct fault injection and, if not already done, injected with the fault. This ensures a FI for variables that are not assigned in the current evalution cycle.

[... cut ...] . However, in the VXML cells from generate statement are only availabe through `<netlist<module>..>` traverse and not the hierarchical `<cells<cell - ..>..>`. However, this should be solvable.

Our toolchain is currently not public, however, we would be happy to contribute to Verilator if there is interest.
Verilator already employs some form of AST matching scheme, so the source transformation via LLVM might not be needed in the first place if a similar feature would be introduced to Verilator.

endrikaja1, 2020-12-22:
Please check https://github.com/verilator/verilator/pull/2724 for the source code

wsnyder, 2021-09-08:
Good discussion and thanks for the code... think no remaining issues to discuss at this time.

closed 2021-09-08 by wsnyder