verilator/verilator #2677: Support randc if only as alias to rand

wsnyder, 2020-12-07:
@kbieganski thanks for adding rand/randomize in #2607 

Do you think you'll support randc soon?  If not can you please submit a patch to record usages of randc and alias them to rand?  Better than just ignoring randc.  If you'll start and finish on randc before release (early January) we can wait for that.

I did some looking at randc and for a real implementation it looks like most simulators support at least 32-bits, so they aren't using a stack of values they shuffle, so they must be using some seeded algorithm.  One implementation would be to have a class instantiated per randc that generates values along these lines:

```
m_nrandLeft [bits-1:0];     // Number of random() calls left before reseed
m_adder [bits-1:0];  // Random addition to value
m_flipper [bits-1:0];   // Random bits to flip
m_rotate [log2(bits)-1:0];   // Rotation of

random_value() {
  if (VL_UNLIKELY(!m_nrandLeft--)) {
    m_nrandLeft = VL_MASK_Q(bits);
    m_adder = random() & VL_MASK_Q(bits);
    m_flipper = random() & VL_MASK_Q(bits);
    m_rotate = random() % bits;
  }
  ++m_adder;
  return VL_MASK_Q(bits) & rotate(m_adder ^ m_flipper, m_rotate);
}
```

This won't be great randomization but should be good enough and reasonably fast.

kbieganski, 2020-12-08:
`randc` is highly unlikely this year, so I submitted #2680.

wsnyder, 2020-12-10:
Thanks for the updates.

closed 2020-12-10 by wsnyder