verilator/verilator #2679: SV/Verilog loop unroll strategy question

logicmonkey, 2020-12-08:
This relates to an [earlier query](https://github.com/verilator/verilator/issues/2678) where I made my test case too simple - although the answer was still very useful!

I have an arithmetic shift right. I iterate to generate log2(N-1) levels of logic for an N bit input. There are N levels of wiring though. Each logic level is a N x 2:1 muxes. The loop variable simply wires those muxes together (you can make a barrel shifter really easily like this).

The coding style is intended to limit the synthesis tool's options. (Of course I know >>> exists :+1: )

The code on the left works but requires the lint waivers. The code on the right doesn't compile in verilator despite the unpacked array.

So does verilator not unroll loops in the way other tools elaborate the RTL?

Just curious - any comment gratefully received!

```
  1 module asr(y, a, s);                                                    |  1 module asr(y, a, s);                                                     
  2   parameter integer                                 BITS = 32;          |  2   parameter integer                                 BITS = 32;
  3   localparam integer                              L2BITS = $clog2(BITS);|  3   localparam integer                              L2BITS = $clog2(BITS);
  4                                                                         |  4 
  5   output logic                                [BITS-1:0] y;             |  5   output logic                                [BITS-1:0] y;
  6   input  wire                                 [BITS-1:0] a;             |  6   input  wire                                 [BITS-1:0] a;
  7   input  wire                               [L2BITS-1:0] s;             |  7   input  wire                               [L2BITS-1:0] s;
  8                                                                         |  8 
  9   // verilator lint_off UNOPTFLAT                                       |  9   logic                                       [BITS-1:0] t [L2BITS:0];
 10   wire                              [L2BITS:0][BITS-1:0] t;             | 10 
 11   // verilator lint_on UNOPTFLAT                                        | 11   always_comb begin
 12                                                                         | 12     t[0] = a;
 13   assign t[0] = a;                                                      | 13 
 14                                                                         | 14     for (int nn=0; nn < L2BITS; nn++)
 15   for (genvar nn=0; nn < L2BITS; nn++)                                  | 15       t[nn+1] = s[nn] ? {{(2**nn){t[nn][BITS-1]}},
 16     assign t[nn+1] = s[nn] ? {{(2**nn){t[nn][BITS-1]}},                 | 16                                   t[nn][BITS-1:2**nn]} : t[nn];
 17                                        t[nn][BITS-1:2**nn]} : t[nn];    | 17     y = t[L2BITS];
 18   assign y = t[L2BITS];                                                 | 18   end
 19                                                                         | 19 
 20 endmodule                                                               | 20 endmodule

```

wsnyder, 2020-12-08:
Replication counts are resolved at elaboration time - IEEE says they must be constant expressions (e.g. a genvar not an int), so if a simulator accepts that it is not compliant.

Is there a reason why you don't just use your left case with verilator split_var?

wsnyder, 2020-12-08:
Feel free to followup - but base question answered I believe.

closed 2020-12-08 by wsnyder