verilator/verilator #2684: Casts from packed structs to enums are now unsupported

rswarbrick, 2020-12-10:
This came up in the OpenTitan repository, and I can try to make a minimal test-case if it's not just a case of "rewrite your SV code; you're doing it wrong"!

In that repository, we instantiate a module with a cast on one of the ports:
```systemverilog
  keymgr_sideload_key_ctrl u_sideload_ctrl (
    .clk_i,
    .rst_ni,

    // more ports here ...

    .dest_sel_i(keymgr_key_dest_e'(reg2hw.control.dest_sel)),

    // more ports here ...

  );
```
Here, `reg2hw.control.dest_sel` is a packed structure, defined like this:
```systemverilog
  typedef struct packed {

    // ... more members here...

    struct packed {
      logic [1:0]  q;
    } dest_sel;
  } keymgr_reg2hw_control_reg_t;
```
and `kymgr_key_dest_e` is an enum type:
```systemverilog
  typedef enum logic [1:0] {
    None   = 0,
    Aes    = 1,
    Hmac   = 2,
    Kmac   = 3
  } keymgr_key_dest_e;
```

With Verilator 4.105, this compiles happily, but with the current head of master it spits out the following error:
```
%Error-UNSUPPORTED: ../src/lowrisc_ip_keymgr_0.1/rtl/keymgr.sv:333:17: Unsupported: static cast to 'ENUMDTYPE 'keymgr_pkg::keymgr_key_dest_e'' from 'STRUCTDT$
PE'
                                                                     : ... In instance top_earlgrey_verilator.top_earlgrey.u_keymgr
  333 |     .dest_sel_i(keymgr_key_dest_e'(reg2hw.control.dest_sel)),
      |                 ^~~~~~~~~~~~~~~~~      
```

Looking at the code in `V3Width.cpp`, I think this was probably from the changes in 74ef35d3b, which added support for dynamic casts.

I *think* this cast should be allowed, in which case this is a regression. If I've misunderstood something, please let me know.

wsnyder, 2020-12-10:
This should be allowed, as you suspected the more complicated dynamic casts required a bunch more checking which seem oversensitive, I'll take a look.

wsnyder, 2020-12-10:
Should be fixed, please give it a try and reopen if it's still botched.  Thanks for your good bug report BTW.

closed 2020-12-10 by wsnyder

rswarbrick, 2020-12-11:
Wow, that was quick! I've just checked and this works, thanks.