verilator/verilator #2686: When numbering unnamed generate blocks (genblk<n>) Verilog LRM is not being followed correctly.

tinshark, 2020-12-10:
Tested on: 4.107 devel rev v4.106-54-gb15dd3f8 (master branch: commit b15dd3f)

As per the LRM each generate-construct (both named and unnamed) in a given scope is assigned a number. The generated names are assigned to the unnamed blocks _as per this assigned number_.

In Verilator, we see that the number for these unnamed generate blocks don't match the assigned number when there are some named blocks in between.

Relevant part of LRM below. Emphasis mine.

LRM 1800-2012

> 27.6 External names for unnamed generate blocks
>  
> Each generate construct in a given scope is assigned a number. The number will be 1 for the construct that appears textually first in that scope and will increase by 1 for **each subsequent generate construct** in that scope. All unnamed generate blocks will be given the name “genblk<n>” where **<n> is the number assigned to its enclosing generate construct**.

Code sample to demonstrate this issue:
```
// Example to show the genblk naming for signals
module top;
logic [3:0] dataIn = 4'b1010;
// 1
for (genvar i = 0; (i < 8); i++) begin // should be genblk1
    logic [3:0] sig1;
    assign sig1 = dataIn;
end
// 2
for (genvar i = 0; (i < 8); i++) begin // should be genblk2
    logic [3:0] sig2;
    assign sig2 = dataIn;
end
// 3
for (genvar i = 0; (i < 8); i++) begin : Custom3 // should be Custom3
    logic [3:0] sig3;
    assign sig3 = dataIn;
end
// 4
for (genvar i = 0; (i < 8); i++) begin // should be genblk4 and NOT genblk3
    logic [3:0] sig4;
    assign sig4 = dataIn;
end
endmodule // End Top Module
```
In the above example, the generated name for the last block should be genblk4. In Verilator we see it as genblk3.

wsnyder, 2020-12-10:
This naming wasn't specified in older IEEE versions, and the different simulators all disagreed, so I would recommend writing Verilog code that only uses named blocks.  Regardless this should be cleaned up.

Verilator does do elaboration a bit unusually so it's possible there's a dragon lurking in this, but might be an easy change.

Perhaps you could make a pull request to fix this?  The existing tests might be enough, but if none change then please add one.

tinshark, 2020-12-11:
Thanks for the response and confirming the issue, Wilson. 

I agree that it can be risky to use dotted reference with unnamed blocks, especially when using results from 2 different parsers, which is what I'm trying to do. 

I'm comparing the results of 2 parsers (one from a student license of a commercial parser) and the other being Verilator and am trying to get the names to match.

Regarding a fix for this, I'm afraid my C++/Software skills aren't up there to attempt a fix. It will be immensely helpful if this can be fixed in the parser. I'll be happy to test and verify a patch at my end and provide feedback.

wsnyder, 2020-12-12:
This was somewhat painful as IEEE doesn't define what is a "block". Different simulators make different assumptions, for example if an empty begin/end should count as a block or not.  

Verilator certainly was breaking some cases that everyone else agreed on, so that is resolved, as to the ambiguity remaining be cautioned.

Note to others this might break "genvar" references relying on old Verilator naming, but as said above you really shouldn't be relying on this naming anyways ;)

closed 2020-12-12 by wsnyder

tinshark, 2020-12-17:
I was able to test this fix on a simple example as well as a moderately sized design and I can confirm that the results I see with Verilator and other simulators are the same. Thanks a lot for the response and providing a fix so quickly. This was very helpful!

I will try this on other designs shortly and report back if anything is off.

I agree that relying on the hierarchical name for genvar signals is risky.

alaindargelas, 2021-08-16:
Hi, back to this issue, I'm trying to have Surelog match Verilator's hierarchy, and the fact that Verilator does not follow the IEEE example given in 27.6 does not help at all... Actually Verilator barf on this example from the standard for name collision issues.

```
27.6 External names for unnamed generate blocks
Although an unnamed generate block has no name that can be used in a hierarchical name, it needs to have a
name by which external interfaces can refer to it. A name will be assigned for this purpose to each unnamed
generate block as described in the next paragraph.
Authorized licensed use limited to: NOAA Boulder Labs Library. Downloaded on September 18,2018 at 23:08:29 UTC from IEEE Xplore. Restrictions apply. 
IEEE Std 1800-2017
IEEE Standard for SystemVerilog—Unified Hardware Design, Specification, and Verification Language
796
Copyright © 2018 IEEE. All rights reserved.
Each generate construct in a given scope is assigned a number. The number will be 1 for the construct that
appears textually first in that scope and will increase by 1 for each subsequent generate construct in that
scope. All unnamed generate blocks will be given the name “genblk<n>” where <n> is the number
assigned to its enclosing generate construct. If such a name would conflict with an explicitly declared name,
then leading zeros are added in front of the number until the name does not conflict.
NOTE—Each generate construct is assigned its number as described in the previous paragraph even if it does not
contain any unnamed generate blocks.
For example:
module top;
parameter genblk2 = 0;
genvar i;
// The following generate block is implicitly named genblk1
if (genblk2) logic a; // top.genblk1.a
else logic b; // top.genblk1.b
// The following generate block is implicitly named genblk02
// as genblk2 is already a declared identifier
if (genblk2) logic a; // top.genblk02.a
else logic b; // top.genblk02.b
// The following generate block would have been named genblk3
// but is explicitly named g1
for (i = 0; i < 1; i = i + 1) begin : g1 // block name
// The following generate block is implicitly named genblk1
// as the first nested scope inside g1
if (1) logic a; // top.g1[0].genblk1.a
end
// The following generate block is implicitly named genblk4 since
// it belongs to the fourth generate construct in scope "top".
// The previous generate block would have been
// named genblk3 if it had not been explicitly named g1
for (i = 0; i < 1; i = i + 1)
// The following generate block is implicitly named genblk1
// as the first nested generate block in genblk4
if (1) logic a; // top.genblk4[0].genblk1.a
// The following generate block is implicitly named genblk5
if (1) logic a; // top.genblk5.a
endmodule
```

wsnyder, 2021-08-16:
@alaindargelas if there's additional fixes needed, which it seems there is, please open a new issue (as this one was closed long ago), and ideally perhaps you'd be willing to look at a pull request to fix it?  thanks.