verilator/verilator #2690: Feature Request: Add support for wires being reals instead of bits

PeterMonsson, 2020-12-11:
Thanks for taking the time to report this.

What would you like added/supported?
I have seen big EDA simulators support this concept where a wire gets turned into a real value through the verilog hierarchy. This is specifically to support behavioural models inside netlists extracted from the analog schematic entry tool. The netlist will itself connect up modules with wires.  

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
I am not sure that this runs on other simulators, but it shows the concept:

    // DESCRIPTION: Verilator: Verilog Test module
    //
    // This file ONLY is placed into the Public Domain, for any use,
    // without warranty, 2020 by Peter Monsson.
    // SPDX-License-Identifier: CC0-1.0
    
    module t (/*AUTOARG*/
       // Inputs
       clk
       );
    
       input clk;
       integer cyc; initial cyc=1;
    
       wire    gain = 1'b0;
       real    in;
       always_comb in = (cyc-4) * 1.0;
       wire    cmp;
       
       adc_netlist netlist(clk, in, gain, cmp);
       
       always @ (posedge clk) begin
          if (cyc!=0) begin
             cyc <= cyc + 1;
             $display("cyc=%0d cmp=%d", cyc, cmp);
             if (cyc==10) begin
                $write("*-* All Finished *-*\n");
                $finish;
             end
          end
       end
    
    endmodule
    
    // Analog Frontend ADC example
    // input, PGA, comparator. 
    
    module adc_netlist(clk, in, gain, cmp);
       input clk;
       input in;
       input gain;
       output cmp;
       
       // real in; //Fails in verilator without this, but not in Big EDA simulator
       
       wire pga_out; //TODO: convert to real or support real 
       // can also be an array of wire in analog netlist: wire [1:0]
       pga_model pga0(in, gain, pga_out);
       comparator_model cmp0(clk, pga_out, cmp);
       
    endmodule
    
    module pga_model(in, gain, out);
       input in;
       input gain;
       output out;
    
       real in;
       real out;
       
       always_comb begin
          out = in * 3.0;
       end
       
    endmodule
    
    module comparator_model(clk, in, cmp);
       input clk;
       input in;
       output cmp;
    
       real in;
       
       reg cmp;
       
       always_ff @(posedge clk) begin
          cmp <= in > 0.0;
       end
       
    endmodule
    

May we assist you in trying to fix this yourself?

Sure, but I have no idea where to start. I am also not sure how long this would take. Finally, is this something that you would like to see supported? I don't know if this is even valid according to the spec.

wsnyder, 2020-12-12:
There's a Verilator issue here but not the one you asked about ;)

Running your code I get a gcc error "error: lvlue required as left operand of assignment"
This is because you are attaching a to an integer at the level above.  This should either work, or give an error.

It is not correct that the problem is the wire should automatically become a "wire real".  IEEE 1800-2017 6.7.1 says: "If a data type is not specified in the net declaration or if only a range and/or signing is specified, then the data type of the net is implicitly declared as logic.

Verilator is doing that correctly, if you manually add a new wire (that is integer) and connect it to your real pin Verilator gives the same results as other simulators.  If in contrast you use "wire real" it mismatches other simulators.

So action is to either support, or error, or warning out on such real-to-integer misconnects.

wsnyder, 2020-12-13:
Added error. After #2675 and general data type conversion gets cleaned up this will become a conversion warning.

closed 2020-12-13 by wsnyder

PeterMonsson, 2020-12-13:
Thank you for the quick resolution!

Going back to a 30,000 feet view: I want to turn some wires into reals throughout the hierarchy. Does verilator or another tool from veripool facilitate this in some way?

wsnyder, 2020-12-13:
If you want XML you add -Wno-UNSUPPORTED --xml  you should be able to get a xml file, then look at the pin's data type compared to what connects to it below's data type.  If you're just doing this once, just look at the new error messages spit out, then correct what Verilator complains about, might need a few passes if the signal goes up several levels.

PeterMonsson, 2020-12-15:
Thank you for the guidelines