verilator/verilator #2696: VPI: vpiLeftRange always returns higher value, vpiRightRange lower value

garmin-mjames, 2020-12-13:
For unpacked arrays, `vpi_handle(vpiLeftRange, array_hdl)` always returns the higher value, no matter if the **msb** is greater or less than the **lsb**. `vpi_handle(vpiRightRange, array_hdl)` always returns the lower value.

This modified **t_vpi_var.cpp** test fails:
```diff
diff --git a/test_regress/t/t_vpi_var.cpp b/test_regress/t/t_vpi_var.cpp
index f49e4b16..3da9a3d0 100644
--- a/test_regress/t/t_vpi_var.cpp
+++ b/test_regress/t/t_vpi_var.cpp
@@ -311,6 +311,25 @@ int _mon_check_var() {
         CHECK_RESULT_CSTR(p, "vpiConstant");
     }

+    TestVpiHandle vh5 = VPI_HANDLE("quads");
+    CHECK_RESULT_NZ(vh5);
+    {
+        TestVpiHandle vh10 = vpi_handle(vpiLeftRange, vh5);
+        CHECK_RESULT_NZ(vh10);
+        vpi_get_value(vh10, &tmpValue);
+        CHECK_RESULT(tmpValue.value.integer, 2);
+        p = vpi_get_str(vpiType, vh10);
+        CHECK_RESULT_CSTR(p, "vpiConstant");
+    }
+    {
+        TestVpiHandle vh10 = vpi_handle(vpiRightRange, vh5);
+        CHECK_RESULT_NZ(vh10);
+        vpi_get_value(vh10, &tmpValue);
+        CHECK_RESULT(tmpValue.value.integer, 3);
+        p = vpi_get_str(vpiType, vh10);
+        CHECK_RESULT_CSTR(p, "vpiConstant");
+    }
+
     return 0;
 }

diff --git a/test_regress/t/t_vpi_var.v b/test_regress/t/t_vpi_var.v
index 34b94b34..fc7d3b81 100644
--- a/test_regress/t/t_vpi_var.v
+++ b/test_regress/t/t_vpi_var.v
@@ -29,7 +29,7 @@ extern "C" int mon_check();
    reg [2:1]   twoone          /*verilator public_flat_rw @(posedge clk) */;
    reg [2:1]   fourthreetwoone[4:3] /*verilator public_flat_rw @(posedge clk) */;

-   reg [61:0]  quads[3:2]      /*verilator public_flat_rw @(posedge clk) */;
+   reg [61:0]  quads[2:3]      /*verilator public_flat_rw @(posedge clk) */;

    reg [31:0]     count        /*verilator public_flat_rd */;
    reg [31:0]     half_count   /*verilator public_flat_rd */;
```

**What 'verilator --version' are you using?  Did you try it with the git master version?**
```
Verilator 4.107 devel rev v4.106-76-ge2e3cc84 (master)
```

**Would you be willing to try to fix Verilator yourself with assistance?**
Yes

wsnyder, 2020-12-13:
Not surprising. Thanks for looking at making a pull. Please be sure to be on up-to-date master as much of the more-general base class hi/lo/left/right stuff was audited and cleaned up this week.

closed 2021-01-03 by wsnyder

garmin-mjames, 2021-01-04:
@wsnyder 🎉 Thanks for fixing this, I wasn't able to find time.