verilator/verilator #2701: Internals to require verilated_heavy.h

wsnyder, 2020-12-14:
Verilated_heavy.h was originally added only for strings, when GCC took several seconds when a string include was used, and this made a huge compile time impact on tests.  GCC has gotten better, and Verilator now uses verilated_heavy for a lot more than strings, ~65% of the tests use verilated_heavy.  Compiling forcing use of heavy is about 8% compile time on the regressions, but "free" on most large designs as likely already using heavy.

I'm committing enabling always using heavy by setting the needHeavy flag always, if there are problems e.g. a large compile time change we can reconsider.  Assuming not, after some period will remove the related conditionals and cleanup the includes.

closed 2021-06-07 by wsnyder

gezalore, 2021-07-07:
This is marked as fixed, but we still have a verilated_heavy.h and `m_needsHeavy` still defaults to true. What was the intention when closing? If sticking with it, can we plonk verilated_heavy.h at the bottom of verilated.h and get rid of the extra file?

wsnyder, 2021-07-07:
I just pushed removing m_needsHeavy.  I haven't decided on what would be best for includes yet. I was thinking a major restructure - making verilated.h just include other files to encourage cleanliness - then there would be verilated_types.h (most of what's in heavy plus some of the typedefs) and verilated_funcs.h and maybe others.  Part of the reason is so VlWide can be used in the inline functions currently in verilated.h.