verilator/verilator #2703: Enable address sanitizer in CI

yTakatsukasa, 2020-12-15:
#2700 adds --asan option that enables address sanitizer.
As of today, the following 37 tests are failing.
Many of them are memory leaks, but some are heap overflow.
When all tests pass with --asan option, we can enable address sanitizer in CI.

- [ ] t_dpi_import                                                                                                                                                                                                   
- [ ] t_embed1                                                                                                                                                                                                                                       
- [ ] t_hier_block_prot_lib_shared                                                                                                                                                                                    
- [ ] t_prot_lib_shared                                                                                                                                                                                                                           
- [ ] t_scope_map                                                                                                                                                                                                    
- [ ] t_time_vpi_100s10ms                                                                                                                                                                                                                         
- [ ] t_time_vpi_10ms10ns                                                                                                                                                                                            
- [ ] t_time_vpi_1fs1fs                                                                                                                                                                                                                           
- [ ] t_time_vpi_1ms10ns                                                                                                                                                                                             
- [ ] t_time_vpi_1ns1ns                                                                                                                                                                                                                           
- [ ] t_time_vpi_1ps1fs                                                                                                                                                                                              
- [ ] t_time_vpi_1s10ns                                                                                                                     
- [ ] t_time_vpi_1us1ns                                                                                                                                                                                    
- [ ] t_trace_array_fst_threads_1                                                                                                                                                                                              
- [ ] t_trace_array_fst_threads_2                                                                                                           
- [ ] t_trace_c_api                                                                                                                                                                                        
- [ ] t_trace_cat                                                                                                                                                                                                              
- [ ] t_trace_cat_renew                                                                                                                                                                                                                                                      
- [ ] t_trace_cat_reopen                                                                                                                                                                                         
- [ ] t_trace_complex_fst_threads_1                                                                                                                                                                                              
- [ ] t_trace_complex_fst_threads_2                                                                                                                                                                            
- [ ] t_trace_public_func                                                                                                                                                                                                                          
- [ ] t_trace_public_func_vlt                                                                                                                                                                                        
- [ ] t_trace_public_sig                                                                                                                                                                                                                              
- [ ] t_trace_public_sig_vlt                                                                                                                                                                                          
- [ ] t_trace_two_dumpfst_cc                                                                                                                
- [ ] t_trace_two_hdrfst_cc                                                                                                                                                                                      
- [ ] t_trace_two_portfst_cc                                                                                                                                                                                                     
- [ ] t_vpi_cb_iter                                                                                                                         
- [ ] t_vpi_cbs_called                                                                                                                                                                                           
- [ ] t_vpi_get                                                                                                                                                                                                                  
- [ ] t_vpi_get_public_rw_switch                                                                                                                                                                                                                         
- [ ] t_vpi_memory                                                                                                                                                                                                     
- [ ] t_vpi_module                                                                                                                                                                                                                                                   
- [ ] t_vpi_time_cb                                                                                                                                                                                                        
- [ ] t_vpi_var                                                                                                                                                                                                                                       
- [ ] t_vpi_zero_time_cb

wsnyder, 2020-12-16:
Not sure what's up with t_prot_lib_shared, perhaps someone can look?
VPI ones are #2706, will get to this in a few days if someone else doesn't want to.
Other trace ones don't fail on my local system (Ubuntu 20.04), so I will need to research as to why.

yTakatsukasa, 2020-12-16:
I will take a look at t_hier_block_prot_lib_shared and t_prot_lib_shared.

yTakatsukasa, 2020-12-16:
I will check leaks in trace tests.

yTakatsukasa, 2020-12-16:
I will send PR to fix  traces of non-fst  cases and t_flag series.

It looks sanitizer reports more errors if  -O0 is used compared to -Os.
(I locally edit verilated.mk to `-O0 -ggdb` to see easier backrace).

yTakatsukasa, 2020-12-16:
Let me take a look at fst traces.

yTakatsukasa, 2020-12-17:
It looks some of the remaining tests are because asan does not understand free list of vpi objects.
I will try fixing them if no one is working yet.

wsnyder, 2020-12-18:
I almost have VPI all clean, they are related to #2706 and some related bugs inside verilated_vpi and the tests.

wsnyder, 2020-12-19:
VPI clean, might be all good, give it a go.

yTakatsukasa, 2020-12-19:
OK. I will test locally and send a PR to enable --sanitize in CI.

yTakatsukasa, 2020-12-19:
Great! I confirmed that neither gcc nor clang complained.

PR #2715 (2020-12-19): CI:Enable address sanitizer for vlt tests.
Closes #2703
closed 2020-12-19 by wsnyder