verilator/verilator #2711: Q&A: How to enforce local including

Xyp, 2020-12-16:
Hi!

I am verilating a very big design (>200 .v files).
Some modules include files using 'include. The included files always are located where the actual module is located.
However, verilator seems to include those includes not from the same directory, but from the first spot, where the same filename was included.

Hard to describe this properly, maybe let me try to describe it another time in a different way. I have the following file scructure:

DIRECTORY_A
    verilogfile_a.sv  // this one contains 'include test.v
    test.v
DIRECTORY_B
    verilogfile_b.sv // this one contains 'include test.v
    test.v

The 2 test.v files have different content (only defines, no modules).

In the verilator file I pass as +incdir DIRECTORY_A and DIRECTORY_B. Can I somehow isolate those incdir definitions, so that verilator knows from where to pick test.v? Or is it possible to define the current directory to have the highest priority?

I might help myself by renaming the include files to a unique name also, but this might create other issues.

I am quite confident there is a standard solution present for this problem, but I could not spot from the manual what I could do.

Best regards,

Kai
PS: The contributor agreement link does not seem to be correct - it returns 404 error: https://github.com/verilator/verilator/.github/blob/master/CONTRIBUTING.adoc

wsnyder, 2020-12-16:
>I am verilating a very big design (>200 .v files).

FWIW I'd put "big" at over 1000, and very big at 10K+ ;)

>Some modules include files using 'include. The included files always are located
>where the actual module is located.
>However, verilator seems to include those includes not from the same directory, but
>from the first spot, where the same filename was included.

I think I understand your point. There's no way currently to specify that an include should be relative to the requesting file.  If a few other tools handle this in a way that works for this we'd consider a pull request to fix/improve this.

>I might help myself by renaming the include files to a unique name also, but this
>might create other issues.

I would strongly suggest that if you can, to do this.  One project had a duplicate name cause a chip bug because synthesis used different lookup rules and the wrong version was read because names were identical.  See Verilog-Perl's "vrename", and/or Verilog::EditFiles which can add a IP-specific prefix to filenames easily.

>PS: The contributor agreement link does not seem to be correct - it returns 404
>error: https://github.com/verilator/verilator/.github/blob/master/CONTRIBUTING.adoc

Fixed, thanks, it's https://github.com/verilator/verilator/blob/master/docs/CONTRIBUTING.adoc

Closing as have answer - feel free to ask followups.

closed 2020-12-16 by wsnyder

Xyp, 2020-12-18:
I just counted files (never did before), let's call it still big, not very big indeed, allthough it is not far from 10K :-)

Thanks for the quick answer!
I see, that for the original toolchain there are statements like:

-makelib libname1
+incdir+directoryname
modulefilename.v
anothermodule.v
-endlib

-makelib libname2
+incdir+directoryname2
modulefilename2.v
anothermodule2.v
-endlib

I sense, that this is a mechanism to bundle things together in a library, creating implicit isolation between the search directories.

Something similar is not there, right? Will help myself now by renaming and look into the proposed Perl scripts to support.

wsnyder, 2020-12-18:
Right, Verilator does not currently have libraries.