verilator/verilator #2712: Support for package imports in interface declarations

rswarbrick, 2020-12-17:
IEEE 1800-2017 allows package imports as part of an interface declaration. Verilator currently supports things like this:
```systemverilog
interface my_if
#(
  parameter my_pkg::type_e SomeParam = my_pkg::default_val
)(
  ...
  input logic [my_pkg::WIDTH-1:0] my_input,
  ...
);
```
(assuming that `type_e`, `default_val` and `WIDTH` are defined appropriately in `my_pkg`). However, it doesn't support this syntax:
```systemverilog
interface my_if
  import my_pkg::*;
#(
  parameter type_e SomeParam = default_val
)(
  ...
  input logic [WIDTH-1:0] my_input,
  ...
);
```
(it gives a syntax error: the code in `verilog.y` doesn't allow this syntax at all).

Verilator does currently support the equivalent syntax for imports as part of module definitions, so hopefully this wouldn't be a massive job.

wsnyder, 2020-12-17:
Surprised it doesn't just work.  Perhaps just a verilog.y fix as you suggest plus a test?

Would you be able to attempt a pull request to fix this?  Thanks.

jameshanlon, 2020-12-17:
Strangely this issue also separately came up for me.

I've opened a pull request with a fix: https://github.com/verilator/verilator/pull/2714

rswarbrick, 2020-12-17:
Wow, neat, thanks!

wsnyder, 2020-12-17:
Closed with #2714, thanks all.

closed 2020-12-17 by wsnyder