verilator/verilator #2719: Now can not inherit 'VerilatedVcdFile' class anymore. why?

testdrive-profiling-master, 2020-12-20:
On 'Verilator 4.106 2020-12-02'

I found some build broken with "verilated_vcd_c.h".  
This is because 'VerilatedVcdFile' is set to "**final**" class.
So there is no way to override input paramter on 'VerilatedVcdC' class constructor.
`explicit VerilatedVcdC(VerilatedVcdFile* filep = nullptr)`
Was this a necessary?
I need a overriding feature with 'VerilatedVcdFile' to display real-time waveform on GTKWave.
Could you remove this 'final' attribute again?

please check out this class.
https://github.com/testdrive-profiling-master/release/blob/master/include/verilator/verilated_vcd_gtkwave.h

wsnyder, 2020-12-20:
We added final to everything as it makes the code faster but in this case probably makes sense to leave it open.  Could you please submit a pull changing final to VL_NOT_FINAL and make sure that works for you?

testdrive-profiling-master, 2020-12-20:
Yes works well without 'final'. :)
Thanks.

wsnyder, 2020-12-20:
Fixed merged, thanks for your first contribution, hope to get more (next time not triggered by our breaking you of course ;)

closed 2020-12-20 by wsnyder