verilator/verilator #2721: Why is {...}[] unsupported?

mjc0608, 2020-12-21:
The following code does not compile and verilator gives me an error of unsupported syntax:
```verilog
{32'(a) << 32'h2}[7:0];
```
I checked the code and found that the error is injected in V3LinkResolve.cpp:230. I actually found similar pattern (i.e., selection of a  shift operation) in the AST optimized by verilator, and if I comment Line 230-233 verilator can parse the code. Is there any specific reason why this language feature is unsupported?

wsnyder, 2020-12-21:
I believe no one has gotten to it yet.  Perhaps you could submit a pull request to test and implement this?  Be sure to test all the forms ([7], [7:1], [5+:2], [5 -: 2]).

mjc0608, 2020-12-21:
It seems like the parser was already implemented correctly (because I comment these lines in V3LinkResolve.cpp and outputs the XML correctly). Do you want me to implement some test case and comment these lines?

wsnyder, 2020-12-21:
Yes, please test and see what if anything breaks.  The parser does 99% of System Verilog (with some things commented), but later stages past Link might break or give wrong answers which is why it was marked unsupported.

udif, 2021-02-17:
But #2619  ??

wsnyder, 2021-02-18:
Dusted off old test and added some new cases, commenting suggested lines seems to work.  Pushed, please give it a try.

closed 2021-02-18 by wsnyder

mjc0608, 2021-02-18:
Tested on a more complex project. It works. Thanks.

udif, 2021-02-18:
From what I can tell neither Cadence (at least for 20.03) nor Synopsys (checked for me, don't know which version) supports this yet in their simulation tools yet 😃