verilator/verilator #2722: Wire wider than the module parameter doesn't trigger an error

yurivict, 2020-12-22:
The 33-bit wire was accidentally passed as an argument:
```
xmodule m(
	  {~xwire[31], xwire[31:0]}
);
```
for a 32-bit argument:
```
module xmodule(
  arg1
);
input     [31:0] arg1;
```
and no error was generated.

I couldn't find if this should or shouldn't be allowed in Verilog.

But if this situation could be caught by verilator this could save time.

wsnyder, 2020-12-22:
This is legal by IEEE, but regardless when I try it I get a WIDTH warning.

wsnyder, 2020-12-22:
```
module t;
   wire [31:0] xwire;
   xmodule m (
        .arg1({~xwire[31], xwire[31:0]})
        );
endmodule

module xmodule(
  arg1
);
input     [31:0] arg1;
endmodule

```

yurivict, 2020-12-22:
I had ```-Wno-WIDTH``` for other reasons. It indeed prints a warning.

Sorry, and thank you.

closed 2020-12-22 by yurivict