verilator/verilator #2727: Unpacked array of single bit seen as Net in VPI

phhorrein, 2020-12-24:
When using Verilator with cocotb, I have an issue with 2-d unpacked arrays of single bits. It is not seen as an array, but as a Net. 

Issue has been reported in cocotb here (along with another one): https://github.com/cocotb/cocotb/issues/2303

It seems to me that this is a Verilator issue, and I'm able to solve this by simply modifying the generated Vtop__Syms.cpp, increasing the dimension by 1 and setting a 0,0 bound for the first dimension. I could provide a PR doing it in the V3EmitCSyms.cpp, when generating the varInsert calls, but I wanted to get advice on it before.

Do you think this should be fixed?

wsnyder, 2020-12-24:
I would check on master as there's been some fixes, but not sure of this one.  If it's different from other simulators which it sounds it is, a fix would be appreciated.  It's possible it's not a problem in V3EmitCSyms.cpp but that's a good first suspect.

phhorrein, 2020-12-24:
I'll check on master. I already have a patch on this file (it's roughly 5 more lines, which check if unpacked dimension is more than 1, and packed dimension is 1, then it adds 1 to the total of dimensions and add the 0,0 boundaries), but I'm not sure it's the correct way to fix, as it only fixes the interface (unsure if this comes from deeper in the code). 

I'll push my PR to discuss it.