verilator/verilator #2731: %Error: Invalid Option: --build

SZCAl, 2020-12-29:
Currently I'm experiencing an issue with verilator where it does not recognize build as a command. I'm currently following the example in the documentation step by step and for some reason any time the --build command is used it does not recognize it.

This is the current line I'm on in the C++ example. 
### verilator -Wall --cc our.v --exe --build sim_main.cpp ###

I installed Verilator through apt and am running Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1
My make version: GNU Make 4.2.1
GCC version: gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)

I'm using the example testbench and verilog files ours.v and sim_main.cpp

wsnyder, 2020-12-29:
Older versions didn't support --build, please use a newer version and/or refer to "verilator --help" which will be for your installed version, or the examples that [hopefully] came with your installed version. Thanks.

closed 2020-12-29 by wsnyder

SZCAl, 2020-12-29:
Is there a reason that the newest version on package managers is running so far behind?

wsnyder, 2020-12-29:
distros generally freeze tools at date of their prelease, so if on e.g. Ubuntu 18.04 they will be at a 2018 January release forever, though depends on distro. Generally for EDA tools better to install yourself.

IvanVnucec, 2021-06-24:
@wsnyder Is there a way to install newer version of Verilator with apt package manager?

wsnyder, 2021-06-25:
If someone provides an apt build you should be able to point to it, but I do not know if anyone does. An alternative  is to use the docker images.