verilator/verilator #2732: 'WEXITSTATUS' was not declared in this scope

catkira, 2020-12-30:
MSYS2 on Windows 10 64 bit, current stable branch
I get the following error:

g++ -std=gnu++14  -MMD -I. -I.. -I.. -I../../include -I../../include -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Os.cpp -o V3Os.o
../V3Os.cpp: In static member function 'static int V3Os::system(const string&)':
../V3Os.cpp:353:46: error: 'WEXITSTATUS' was not declared in this scope
         const int exit_code = WEXITSTATUS(ret);
                                              ^
../Makefile_obj:292: recipe for target 'V3Os.o' failed
make[2]: *** [V3Os.o] Error 1
make[2]: Leaving directory '/d/_shared_git/verilator/src/obj_dbg'
Makefile:65: recipe for target '../bin/verilator_bin_dbg' failed
make[1]: *** [../bin/verilator_bin_dbg] Error 2
make[1]: Leaving directory '/d/_shared_git/verilator/src'
Makefile:220: recipe for target 'verilator_exe' failed
make: *** [verilator_exe] Error 2

Inserting this in V3Os.cpp fixed it:
# ifndef WEXITSTATUS
#  ifdef __MINGW32__
#   define WEXITSTATUS(w)	(((w) & 0xC0000000) == 0)
#  else
#   define WEXITSTATUS(w)	(((w) >> 8) & 0377)
#  endif
# endif

wsnyder, 2020-12-30:
Thanks for reporting. There's already similar code in V3Os.cpp, did you try the current master?   If you did can you please indicate via a pull request what needs to be changed (as it isn't just and insert), thanks.

catkira, 2020-12-30:
I tried the current master but I could not compile because of msys2 issues I think. I switched to using the WSL2 (windows subsystem for linux 2) now and everything works very fine. I use an Ubuntu 18.04 container and then just do apt-get install verilator.

closed 2020-12-30 by catkira