verilator/verilator #2733: FlexLexer.h: No such file or directory

catkira, 2020-12-30:
MSYS2 on Windows 10 64 bit, current stable branch
I get the following error:

g++ -std=gnu++14  -MMD -I. -I.. -I.. -I../../include -I../../include -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseLex.cpp -o V3ParseLex.o
In file included from ../V3ParseLex.cpp:28:0:
V3Lexer_pregen.yy.cpp:394:23: fatal error: FlexLexer.h: No such file or directory
compilation terminated.
../Makefile_obj:297: recipe for target 'V3ParseLex.o' failed

Solution:
I had to append CXXFLAGS=-I path_to_FlexLexer.h to ./configure (It does not work with FlexLexer 2.6.2 but works with 2.6.0 !)

wsnyder, 2020-12-30:
This would be then a bug in MSYS distro/package that installed flex, not verilator; please consider reporting it to them.  (Assuming you installed flex from a package - if you installed it manually then you would have to point to it.)

Thanks for reporting it here though as someone else with a similar problem then knows how to resolve it.

closed 2020-12-30 by wsnyder

catkira, 2020-12-30:
I installed this package in conda: https://anaconda.org/msys2/m2-flex
For me the file is in /usr/include but the build scripts dont look there.

wsnyder, 2020-12-30:
It's up to the packager to put the file in a place which the compiler has as a default include path, not the user/build script.  /usr/include is pretty standard on Linux so perhaps it's a gcc package problem rather then flex, but either way MSYS's issue.

catkira, 2020-12-30:
yes, g++ -v gives:
$ g++ -v
Using built-in specs.
COLLECT_GCC=C:\ProgramData\Anaconda3\Library\mingw-w64\bin\g++.exe
COLLECT_LTO_WRAPPER=C:/ProgramData/Anaconda3/Library/mingw-w64/bin/../lib/gcc/x86_64-w64-mingw32/5.3.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-5.3.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/x86_64-w64-mingw32/include --libexecdir=/mingw64/lib --with-gxx-include-dir=/mingw64/include/c++/5.3.0 --enable-bootstrap --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,objc,obj-c++,fortran,ada --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-time=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-version-specific-runtime-libs --disable-isl-version-check --enable-lto --enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev5, Built by MSYS2 project' --with-bugurl=https://sourceforge.net/projects/msys2 --with-gnu-as --with-gnu-ld
Thread model: posix
gcc version 5.3.0 (Rev5, Built by MSYS2 project)

so it looks like it is only looking in /mingw64/include/c++/5.3.0  and /mingw64/x86_64-w64-mingw32/include

catkira, 2020-12-30:
I guess its not worth to waste so much time using Windows...

wsnyder, 2020-12-30:
So yes, a flex packaging issue.

Yes, if you can use Linux it's what 99% of Verilog development is done on, so things will generally go better.

catkira, 2020-12-30:
I switched to WSL2 now with ubuntu18 as guest OS. It was very easy to configure and after 15 min everything was running. I recommend that you make this the default recommendation for people who want to use Windows.

cr1901, 2022-06-30:
This is a [wontfix](https://github.com/msys2/MSYS2-packages/issues/1911) upstream. The workaround is to copy `FlexLexer.h` somewhere your compiler is looking for files. I've found that `cp /usr/include/FlexLexer.h /mingw64/include` worked for me to get a `verilator` that builds.