verilator/verilator #2734: Support testing with ivtests in Verilator extended tests?

wsnyder, 2020-12-31:
Consider adding ivtests (https://github.com/steveicarus/ivtest), which is used for Icarus testing, to the Verilator extended tests repository and CI.  @caryr indicates that they'd likely be willing to add Verilator-specific testlists to ivtest so the information could be held in ivtest.

caryr, 2020-12-31:
The first place to start on this is a custom regression script to run Verilator instead of iverilog. This custom script will also take a Verilator specific regression list that will then be used to map features not supported to an appropriate failure category. For example, any test that has delay would likely be mapped to not implemented (NI).

I am willing to help write the custom script and create the Verilator specific regression list. At the moment I do not have Verilator installed and am using most of my free time working on Icarus. I do believe creating a good test base is important and have recently been working on getting ivtest clean in the sv-tests framework by resolving issues in ivtest and categorizing issues that are not part of the standard, but industry accepted (de-facto standard) constructs.

My rational for making this part of ivtest is it provides direct ongoing feedback regarding tests which will allow us to catch and correct misunderstanding quicker. Having an audited Verilator regression list will also allow you to identify missing functionality. It also makes sense that I look at the Verilator test suite and how that could be used in Icarus or does it even make sense right now. The first thing is probably a discussion of the testing strategies used. For example anything that has systemC is currently a no go for Icarus, but maybe a long term goal would be to integrate systemC into Icarus.

My biggest concern regarding getting ivtest to work with Verilator is we extensively use delays to sequence checks. Some tests could be changed to be compatible, but we would need to be careful not to change any test that is specifically testing delay functionality.

Let me know if you have any questions.

wsnyder, 2020-12-31:
Another option to consider when we look into it is to use Verilator's harness which is a lot better for Verilator debugging, and just have it read the test lists somewhat similar to how sv-tests does it.  Then ivtests can just provide the lists and tests and ignore the harness otherwise.

caryr, 2020-12-31:
Yes, that will work as well. For sv-tests I am going to write a script to generate the list of tests because just reading the lists files like it does now is not correct. I will keep Verilator in mind when writing this script so we can use either method. I would still like the Verilator exclusion list to be part of ivtest since that closes the feedback loop regarding test validity.

wsnyder, 2023-08-30:
We've been running ivtests using sv-tests, and expect we'll continue to rather than build our own harness.

closed 2023-08-30 by wsnyder