verilator/verilator #2739: CI: Fix internal coverage

wsnyder, 2021-01-03:
Internal coverage seems broken with recent CI changes, namely it seems to redo the build for some reason (and do it without the coverage flags).  Maybe this is because nodist/code_coverage is starting at stage 3?  We can have it redo the configure but this then breaks picking up the artifacts.

https://github.com/verilator/verilator/runs/1638342661?check_suite_focus=true

@umarcor

umarcor, 2021-01-03:
> Internal coverage seems broken with recent CI changes, namely it seems to redo the build for some reason (and do it without the coverage flags).

Which recent changes do you mean? Can you narrow down the period?

Which coverage flags are missing?

> Maybe this is because nodist/code_coverage is starting at stage 3? We can have it redo the configure but this then breaks picking up the artifacts.

All coverage test jobs start at stage 3 because of:

- https://github.com/verilator/verilator/blob/master/.github/workflows/coverage.yml#L97
- https://github.com/verilator/verilator/blob/master/.github/workflows/coverage.yml#L62-L71
- https://github.com/verilator/verilator/blob/master/ci/ci-script.bash#L99-L127

However, I don't know what does "starting at stage 3" mean.

wsnyder, 2021-01-11:
Fixed last night - needed to persist objects back into obj directory, and reconfigure.

closed 2021-01-11 by wsnyder