verilator/verilator #2741: Modules with reals cannot have coverage collection enabled?

polmacanceart, 2021-01-06:
Take the following module as an example:

`/* verilator coverage_off */
module
   cov_check
     (
      input logic clk1,
      input real delay,
      output logic test1,
      output real test2
      );

   timeunit 1s;
   timeprecision 1ns;

   parameter init_delay = 1e-6;

   always @(posedge clk1)
     begin
	if ($realtime() > init_delay/10)
	  begin
	     test1 <= 'b1;
	     test2 <= init_delay;
	  end
	else
	  begin
	     test1 <= 'b0;
	     test2 <= init_delay + delay;
	  end
     end // always @ (posedge clk1)
   
endmodule // cov_check
/* verilator coverage_on */
`

It compiles cleanly with:

`verilator -Wall -cc cov_check.sv -build --coverage`

However, if the coverage on/off comments are removed:

`In file included from Vcov_check__ALL.cpp:3:0:
Vcov_check.cpp: In static member function ‘static void Vcov_check::_combo__TOP__1(Vcov_check__Syms*)’:
Vcov_check.cpp:75:24: error: invalid operands of types ‘double’ and ‘double’ to binary ‘operator^’
     if ((vlTOPp->delay ^ vlTOPp->top__DOT____Vtogcov__delay)) {
          ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vcov_check.cpp: In static member function ‘static void Vcov_check::_sequent__TOP__3(Vcov_check__Syms*)’:
Vcov_check.cpp:93:24: error: invalid operands of types ‘double’ and ‘double’ to binary ‘operator^’
     if ((vlTOPp->test2 ^ vlTOPp->top__DOT____Vtogcov__test2)) {
          ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Vcov_check__ALL.cpp:4:0:
Vcov_check__Slow.cpp: In static member function ‘static void Vcov_check::_settle__TOP__2(Vcov_check__Syms*)’:
Vcov_check__Slow.cpp:51:24: error: invalid operands of types ‘double’ and ‘double’ to binary ‘operator^’
     if ((vlTOPp->delay ^ vlTOPp->top__DOT____Vtogcov__delay)) {
          ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vcov_check__Slow.cpp:59:24: error: invalid operands of types ‘double’ and ‘double’ to binary ‘operator^’
     if ((vlTOPp->test2 ^ vlTOPp->top__DOT____Vtogcov__test2)) {
          ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [Vcov_check__ALL.o] Error 1
`
I am using Verilator 4.104

Shouldn't the reals be simply ignored from a coverage point of view?

wsnyder, 2021-01-06:
Yes, these should presumably be suppressed.  Perhaps you'd be willing to submit a pull request that adds a real to an existing test_regress to show the issue, then fixes it in V3Coverage.cpp?

polmacanceart, 2021-01-10:
The decision to cover or not cover seems to be made in V3AstNodes.h:

    bool isToggleCoverable() const {
        return ((isIO() || isSignal())
                && (isIO() || isBitLogic())
                // Wrapper would otherwise duplicate wrapped module's coverage
                && !isSc() && !isPrimaryIO() && !isConst());
    }

Adding "&& !isDouble()" should fix it?

In my example isIO() will be true, I can add real within the module and it will not generate an error.

I'll try this and see about modifying an existing test_regress

polmacanceart, 2021-01-10:
Branch fix_issue_2741 created and pull request submitted

wsnyder, 2021-01-10:
Fixed and thanks for your pull.

closed 2021-01-10 by wsnyder