verilator/verilator #2742: Verilator doesn't compile with flex 2.6.4 but does with 2.5.37

sjalloq, 2021-01-07:
Hi there,

I'm trying to compile the v4.106 tag in order to get cocotb working and am hitting a compilation error in V3Lexer.  This seems to be due to the flex version and having recently installed the latest flex version, it prevents Verilator compilation.  I know this might well fall under the "it's a problem with your system" banner but wanted to see if you had any input. What version of flex do you recommend.  Your prerequisites don't say anything about flex versions.

I'm on a RHE7.9 machine with GCC 9.3.0.

```
make[2]: Entering directory `/proj/tigris/tools/src/verilator/src/obj_dbg'
      Compile flags:  g++ -MMD -I. -I.. -I.. -I../../include -I../../include -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/proj/tigris/tools/verilator/v4.106/share/verilator"
g++   -MMD -I. -I.. -I.. -I../../include -I../../include -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/proj/tigris/tools/verilator/v4.106/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseLex.cpp -o V3ParseLex.o
In file included from ../V3ParseLex.cpp:28:
V3Lexer_pregen.yy.cpp: In member function ‘virtual int V3LexerBase::yylex()’:
V3Lexer_pregen.yy.cpp:4243:9: error: request for member ‘rdbuf’ in ‘((V3LexerBase*)this)->V3LexerBase::yyin’, which is of pointer type ‘std::istream*’ {aka ‘std::basic_istream<char>*’} (maybe you meant to use ‘->’ ?)
 4243 |    yyin.rdbuf(std::cin.rdbuf());
      |         ^~~~~
```

wsnyder, 2021-01-07:
Ubuntu 20.04 used flex 2.6.4 with GCC 9.3.0, with complete success, so it's something more subtle related to redhat.

Note there's Verilator's src/flexfix which makes a bunch of flex source edits to hack around these sort of issues, so if you figure out a fix, flexfix can be modified to work around the flex/gcc package bug.

sjalloq, 2021-01-08:
OK thanks, I have no idea where to start in debugging these sort of issues so I'll stick to a version that works.

closed 2021-01-08 by sjalloq