verilator/verilator #2743: Exceptions to inlining with --flatten

jameshanlon, 2021-01-07:
There are currently several situations where inlining can't take place, regardless of the inlining multipler:
- Modules with interfaces
- Public modules
- Modules with classes
- Modules annotated with NO_INLINE_MODULE pragma

Rather than inlining being an optimisation for simulation, the intended behavour of `--flatten` to force inlining with `--xml-only` was to produce a flattened 'netlist' of the design, for analysis by downstream tools. If inlining can't take place a flat netlist cannot be produced.

I'm curious about what's involved in forcing inlining for the above exceptions, and if are there any reasons why they prohibitivly difficult to do? Presumably, choosing to ignore public or NO_INLINE_MODULE is straightforward, but I'm not so sure about how interfaces and classes interact with the elaboration flow.

Cheers,
Jamie

wsnyder, 2021-01-10:
The inlining hit bugs with classes and interfaces.  There's some work to redo a bunch of the inliner steps in #1539 which will remove several/all of these problems.

However for public modules and no_inline_module it should be fine to ignore those and flatten with --flatten.

jameshanlon, 2021-01-11:
Thanks @wsnyder, that's usefult to know. In the mean time, I'll create a patch to override public modules and no_inline_modules with --flatten.

jameshanlon, 2021-01-18:
Here's the pull request: https://github.com/verilator/verilator/pull/2761

wsnyder, 2021-01-18:
Thanks for fixing.

closed 2021-01-18 by wsnyder