verilator/verilator #2746: VPI implementation improvements discussion

phhorrein, 2021-01-10:
I'm interested in working on Verilator's VPI implementation to provide various improvements and fixes. I have a number of ideas I have started (or considered) investigating, but I would like to discuss this, or see if some other points have higher priority (or should never be implemented). I'm still new to Verilator internals, and while I use (System)Verilog and simulation tools on a daily basis, I have little insight on how they are working internally, so if any idea seems strange, no problem!

The first point is that current VPI implementation is not 1800-2017 compliant. Adding the sv_vpi_user.h and vpi_compatibility.h file from the standard is a a start, and a number of elements are missing (currently working on this). I'm wondering if compatiblity through the VPI_COMPATIBILITY_VERSION_XXXXvXXXX should be implemented (at least 1 for 1364 and 1 for 1800?). 

The second point is, shouldn't all vpi_ functions be implemented as the vpi_scan function ? Meaning, delegate all operation to the VerilatedVpioXXX objects, and keep the vpi_xxx functions as shell functions only. I don't know what the performance impact would be, it would probably be a little more efficient but I'm not sure. But it feels like it would make the code clearer. 

The third point is more complex and unelaborated, but couldn't the tool benefit from the C++ conversion to compile the VPI + verilated code as a whole, improving the global performance, instead of using the same "vpi" library strategy as other simulators? I was also wondering if it wouldn't be better to provide a VPI-based version of VerilatedVar (through the varInsert function), still with the idea of increasing the links between VPI and design. 

Do you have any opinion about this?

wsnyder, 2021-01-10:
Glad to have more work on VPI!

>Adding the sv_vpi_user.h and vpi_compatibility.h file from the standard is a a start, and a number of elements are missing (currently working on this).

Great!  We want to keep the internal code coverage up, so I'd suggest a first pull that adds them, plus adds a test that checks all of the functions involved just give unimplemented errors.  Then later pulls can implement the real code.

> I'm wondering if compatiblity through the VPI_COMPATIBILITY_VERSION_XXXXvXXXX should be implemented (at least 1 for 1364 and 1 for 1800?).

I'd prefer we only support 1800, as enough work as it is, and 1800 is now 15 years old so well over due for people to port their VPI code.

>The second point is, shouldn't all vpi_ functions be implemented as the vpi_scan function ? Meaning, delegate all operation to the VerilatedVpioXXX objects, and keep the vpi_xxx functions as shell functions only.

If the only cost is a case statement or so then this might be reasonable, but I'm not sure if that's the case.  Also huge functions are hard to maintain/debug so perhaps the opposite of having vpi_scan plus vpi_* call some internal functions is cleaner? Would need to see what it looks like to understand what's better.

>The third point is more complex and unelaborated, but couldn't the tool benefit from the C++ conversion to compile the VPI + verilated code as a whole, improving the global performance, instead of using the same "vpi" library strategy as other simulators? 

We wouldn't want to break VPI compatibility.  Also hopefully VPI isn't a large percentage of runtime, or the user has done something unfortunate and perhaps should do something differently (e.g. DPI).  Finally note the compiler can do pretty well if the user passes all of their code + all verilated code into the same compiler run, though this isn't the way the Verilator makefiles are setup, the user would need their own.

phhorrein, 2021-01-11:
> I'd prefer we only support 1800, as enough work as it is, and 1800 is now 15 years old so well over due for people to port their VPI code.
Agreed, and incompatibilities are few, so it's not that difficult to switch. I've started working on test cases. It will take some time to get it right, I'm checking with other commercial tools (even if they are not always good reference). 

> Would need to see what it looks like to understand what's better.
I tried it with vpi_iterate already. In my opinion, it's cleaner, as it keeps a few supported cases only in each class, instead of the whole set of supported operations in a single function. I can push a draft to illustrate this. 

Do you have any performance tests for VPI? I'm curious to know how expensive function calls are. But I agree, breaking VPI compatibility is probably not a good idea. I'm only wondering if something could not be done to benefit from the fact that we are compiling a single language in the end (something more than a single compiler run). Yet it's probably too much work for little to none improvement.

wsnyder, 2021-01-11:
I don't have a performance VPI test, would welcome one getting added.

The likely critical performance section of the VPI is Verilator making the value change callbacks, and then the callback getting the signal value, likewise for timer registering and callbacks, these should be as efficient and fast as reasonable as many of them can occur every cycle.

Iterating the design is unlikely to be called every cycle (in reasonably written code) so doesn't need to be optimal.

phhorrein, 2021-01-11:
Yes it seems logical to optimize value changes (put and get) first, as they should be the operations called more often. I'm curious to know the performance impact, I'll try and add a basic test to check that.

garmin-mjames, 2021-01-14:
I agree that adding `sv_vpi_user.h` is a good idea. Things like `vpiStringVar` for string variables are nice.
In terms of VPI improvements, it would also be nice to have expanded support for scopes, as currently things like generate statements return a type of `vpiScope`, rather than `vpiGenScopeArray`.

phhorrein, 2021-01-15:
I started creating new VPI tests. For now I'm focusing on variables, I'll also add tests for scopes after that. I didn't create a PR (as it's a slow moving work...) for now, I'll do it as soon as I'm satisfied with the number of test cases. 

I'm planning to focus on: variables model (struct, arrays, packed, integer, string, ...) and associated properties, scopes model, callbacks, value setter/getter for now. If anything else seems useful for a first try, tell me, I can add it to the tests.

wsnyder, 2021-03-12:
Think discussion completed.

closed 2021-03-12 by wsnyder