verilator/verilator #2749: Calling system verilog function from C++

KGHVerilator, 2021-01-13:
Hello,
  I have some system verilog functions that I would like to call from C++. Is that possible with verilator?
  
  My use case is I am trying to use Uncle Bob's test driven design methodology for verilog state machines. To that end, I am writing the state machine in the two process form, one very simple sequential process:

always_ff @(posedge axi_clk_in)
begin
    reg_curr <= reg_next;
    if(axi_rst_in)
    begin
        reg_curr <= 0;
        reg_curr.state <= RESET;
    end
end

And one combinatorial process:
always_comb
    reg_next = combinatorial(reg_curr, go_in, axready, data_width_bytes_in,
        burst_length_in, address_in, qos_in, increment_in, combo);

With this all the intelligence is inside stateless function calls (the state is just another parameter to the function).
This allows for a lot of refactoring of the logic.

I would then like to do something in c++ like
go_in = 1
data_width_bytes_in = 4
Vcombinatorial->eval()
EXPECT_EQ(reg_next.valid, 1)

Do I need to make a small module wrapper for combinatorial?

wsnyder, 2021-01-13:
For your example to work you would need to make this into a small module and Verilate it.  The other alternative is to make a DPI export Verilog function that you call from C++, but all the logic needs to be in the function - it can't make time pass in a posedge block.

Closing as answered but feel free to ask follow up questions.

closed 2021-01-13 by wsnyder

KGHVerilator, 2021-01-13:
Thank you wsnyder. That is not the answer I wanted but it does answer my question. I will make a little module.
Kendrick

PS, sorry for the 3 copies of this issue. When I tried to open it, I got an error (twice) the third submit was accepted. Later I saw that three issues were opened.