verilator/verilator #2752: Too many preprocessor tokens on a line

ldoolitt, 2021-01-13:
Regression applying Verilator to production application code, from v4.010 to v4.034
Pretty clearly caused by the fix to issue #1574.
"oddball huge defines" describes my use case exactly.
I understand the need for a limit, halting problem and all that. Can there be some configuration of the limit?

wsnyder, 2021-01-14:
Wow, didn't think that one would get hit.

What number works for you?  Given it's lasted 2 years perhaps only a small increase is sufficient.  Either way (fixed increase or making it programmable) perhaps you could contribute a pull request?

ldoolitt, 2021-01-14:
LINE_TOKEN_MAX = 24000 is enough for my use case .. for now.  I do think a configurable limit is better.  I'll study the code and see if I can come up with a pull request.

wsnyder, 2021-02-21:
For now increased the limit, glad to consider pull request if you/someone wants to make this programmable.

closed 2021-02-21 by wsnyder