verilator/verilator #2753: Incorrect range inheritance in port declaration

embeddedgo, 2021-01-14:
Verilator 4.109 devel rev v4.108-9-gab5d4bd5

Running `verilator --lint-only --Wall` for the code below:

```systemverilog
`default_nettype none

module pwm #(dutyBits = 2) (
	input
		wire [dutyBits-1:0] duty,
		wire                clk,
	output
		logic               out
);
	logic [dutyBits-1:0] cnt;

	always_ff @(posedge clk) begin
		cnt <= cnt + 1'b1;
	end

	assign out = (cnt < duty);
endmodule

module ledpwm (input wire clk, output wire led);
	localparam dutyBits = 8;

	wire [dutyBits-1:0] duty = 123;

	pwm #(dutyBits) pwm1 (.clk, .duty, .out(led));
endmodule
```

gives the following warning:

```
%Warning-WIDTH: pwm.sv:26:25: Input port connection 'clk' expects 8 bits on the pin connection, but pin connection's VARREF 'clk' generates 1 bits.
                            : ... In instance ledpwm
   26 |  pwm #(dutyBits) pwm1 (.clk, .duty, .out(led));
      |                         ^~~
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Error-UNSUPPORTED: pwm.sv:14:14: Unsupported: Non-single bit wide signal pos/negedge sensitivity: 'ledpwm.__Vcellinp__pwm1__clk'
   14 |  always_ff @(posedge clk) begin
      |              ^~~~~~~
%Error-UNSUPPORTED: pwm.sv:8:24: Unsupported: Clock edge on non-single bit signal: 'ledpwm.__Vcellinp__pwm1__clk'
    8 |   wire                 clk,
      |                        ^~~
%Error: Exiting due to 2 error(s), 1 warning(s)
```

It seems the `clk` port in `pwm` module inherited the width from `duty` port. If you swap the order of `pwm` input ports the lint ends with success.

The *IEEE Std 1800-2017* in *23.2.2.3 Rules for determining port kind, data type, and direction* says:

> For subsequent ports in an ANSI style port list:
  —    If the direction, port kind and data type are all omitted, then they shall be inherited from the previous
       port. If the previous port was an interconnect port, this port shall also be an interconnect port.
  Otherwise:
  —    If the direction is omitted, it shall be inherited from the previous port.
  —    If the port kind is omitted, it shall be determined as previously specified.
  —    If the data type is omitted, it shall default to logic except for interconnect ports that have no
       data type.

The `clk` should be simply `wire logic` not `wire logic [dutyBits-1:0]`.

wsnyder, 2021-01-14:
Agreed, I don't think I've ever seen rel code without an input/output for every signal, so not surprising wasn't tested well.

Perhaps you'd be willing to contribute a pull request to test and fix this?

wsnyder, 2021-01-14:
BTW need to be sure we check both ANSI and non-ANSI forms as code is different.

embeddedgo, 2021-01-14:
I've just started learning SystemVerilog and found this type of declaration in some learning materials. The sv2v tool I use to convert SV to Verilog before run ISE xst is happy with it.

If I understand the IEEE1800-2017 syntax for non-ANSI port declarations correctly, you can't specify multiple port types after a direction keyword. Both sv2v and Verilator return a syntax error for the code below:

```SystemVerilog
module pwm(clk, duty, out);
	input
		wire [dutyBits-1:0] duty,
		wire                clk;
	output
		logic               out;
```

I've no real experiences with Verilator so I'cant help with a parser fix. I've just started to use it as simulator and much more as fast syntax checker (I use a plain text editor to edit code and commandline tools to check/compile it). 

Verilator with --lint-only option is much faster than Xilinx's xst with -compileonly flag and it knows SV directly. In my flow Verilator and sv2v+xst are checking each other.

wsnyder, 2021-02-24:
Thanks for the report - fixed in git towards 4.410.

closed 2021-02-24 by wsnyder