verilator/verilator #2755: Add a single line "lint_ignore" verilator comment

phhorrein, 2021-01-14:
Currently, to selectively disable linting, it's necessary to add a lint_off statement before, and a lint_on statement after. While this is perfectly reasonable for big chunks of code, it is breaking code readability for single lines. 

Would it be possible to add a single line "lint_ignore" comment, applied to the line it is found at only?

For example:

```verilog
wire [7:0] my_acceptable_signal; // verilator lint_ignore WIDTH
```

seems much more readable than:

```verilog
/* verilator lint_off WIDTH */
wire [7:0] my acceptable_signal;
/* verilator lint_on WIDTH */
```

Especially when more than one signals are declared.

wsnyder, 2021-01-15:
Agree this is useful, but there's several internal complications.
- Internally the suppression information is made with each token, so end of line would require backtracking what to correct after-the-fact.
- Some warnings are printed in the parser, and again those would be too late.
- Perhaps the preprocessor could move the comments up front - though that still wouldn't fix any preprocessor warnings.

Would entertain a patch if you/someone want to work on it.

wsnyder, 2021-01-15:
Another possibility would be

// verilator lint_off_next_line  asdlajsdlaksd

which would then turn it on automatically after the next non-comment line.  This would be much simpler, and also help avoid really long lines.  Probably preferable...

phhorrein, 2021-01-15:
lint_off_next_line would be better than lint_off/lint_on for those case, but when declaring a block of related signals, it's still not optimal. I'll try and implement it for a start, and see what could be done for the lint_ignore (even if it seems not that easy from what you tell and what I see in the code).

wsnyder, 2021-11-28:
This was fixed a while ago with #2870.

gezalore, 2021-11-28:
@wsnyder #2870 was never merged

ganoam, 2022-03-29:
If I may add a to this request. It would be useful if the `lint_ignore` comment came with an argument as to what rule I want to be ignored. E.g. ideally I'd want to write stuff like this:
```
counter i_counter (
  .clk_i(clk_i),
  ...
  .empty(), // lint_ignore PINCONNECTEMPTY
  .full(cnt_full),
);
```
Would be cool to see that :)

wsnyder, 2022-03-29:
Since it needs to fit on one line we probably need multiple warnings allowed

   .empty(), // lint_ignore PINCONNECTEMPTY,SOMEOTHERWARNING

wallento, 2022-03-29:
I would suggest putting braces around it for better parsing and ability to use the remain for an explanation etc.

    .empty(), // lint_ignore(PINCONNECTEMPTY,SOMEOTHERWARNING) this is here for historic reasons

wsnyder, 2025-10-08:
This now works:
```
 // verilator lint_off WIDTHTRUNC,WIDTHEXPAND  // after slashes ignored
```

As to a single-line comment in addition to the concerns above, another problem is it may break on any automatic reformatting of the code (e.g. verible-format).