verilator/verilator #2757: Unsupported: Compile multi-threads's model with msvc on windows 10

phantom-killua, 2021-01-16:
> Verilator Version:
```
Verilator 4.100 2020-09-07 rev v4.040-74-g16fba5948
```
> OS and Compiler:
```
Windows 10 and Visual Studio 2017
```
> Verilator Command:
```
verilator -Wall --sc --trace --threads 2 --compiler msvc example.v 
```
> Add all generated source files to the VS project，then compile the project. Unfortunately, the compiler reports the following error:
```
verilated_threads.h(166): error C3861: 'GetCurrentProcessorNumber': identifier not found
```
> I think that this error is caused by the missing header file, then I try the way this link([getprocessidofthread-identifier-not-found](https://stackoverflow.com/questions/30029140/getprocessidofthread-identifier-not-found)) provides. the compiler still reports the same error. I hope you can help me solve this problem, thanks.

> _**Notice:**_ compile and run single-thread model normally.

wsnyder, 2021-01-16:
I presume you are using a recent windows, and the including of processthreadsapi.h didn't help?  Sorry I don't use windows so my source is Googling just as you did, if that doesn't work you might want to post to stack overflow what you have tried.  Obviously when you figure it out please submit a pull request to fix it.

wsnyder, 2021-03-12:
Closing due to age, please reopen if still a problem or post pull request if fix is known.

closed 2021-03-12 by wsnyder