verilator/verilator #2760: Standard library Map rises abort

victorspgl, 2021-01-17:
Hello, 

I am working with lowrisc-chip fork. I have been using verilator with this model for years. But now I am experiencing a weird bug. When I change my verilator version from 4.030 to 4.040 or above (I have tried with all of them) I am experiencing a weird abort when initializing some structures. The function that rises the abort is **memory_model_init** at [dpi_ram_behav](https://github.com/lowRISC/lowrisc-chip/blob/refresh-v0.6/src/test/cxx/common/dpi_ram_behav.cpp). 

I am using gcc and g++ version 10, but it also happens with version 9.

I have used GDB to see the back trace, and what I get is: 

at 0x4C2318B: raise (raise.c:51)
by 0x4C02858: abort (abort.c:79)
by 0x4BA0675: ??? (in /lib/x86_64-linux-gnu/libgcc_s.so.1)
by 0x11146E: memory_model_init (dpi_ram_behav.cpp:131)

At first I thought that maybe the issue was related with thread safe functions, but after trying different thread flags (no-threads, threads 1, etc),  I was not able to avoid the error. I also thought that maybe c++11 support was the problem, but I was using std=c++11 with verilator v4.030

victorspgl, 2021-01-17:
I forgot to mention that I have other partners working with verilator v4.102 with the same model and they do not experience the problem. Therefore, it is not related with our code, but with the libraries or verilator.

wsnyder, 2021-01-17:
It sounds like some sort of memory corruption.  This might only be a side effect of something Verilator changed that modified the layout of your program, though could of course also be a Verilator bug.

Here's some ways to help debug:
- Compiling and linking with debug (-ggdb) so the backtrace is more sensical.
- Compiling and linking with "-O0"
- Compiling and linking with "-fsanitize=address"
- Using 'git bisect' to find the exact verilator commit that is causing the problem

Please let us know how it turns out - I assume your goal is to get up to the most recent release.

victorspgl, 2021-01-18:
Hi, I was already using "-O0 -gdb". 

I have been debugging a little bit and It seems to be a g++ issue. I have been able to isolate the problem using only g++ without verilator and the error was still there. There are two solutions to remove the problem, either by changing my code or using gcc-7.5 

I will close this issue and I will open another in g++

closed 2021-01-18 by victorspgl