verilator/verilator #2763: error due to no-timescale on primitive

zhuzhzh, 2021-01-22:
The simulation needs to read the library sim file which includes the udp.
`timescale is in the library sim file. but verilator can't handle it.
Here is the code to reproduce the issue.

**top.v**

```verilog
`timescale 1ns/1ns
module top;

wire out;
wire in; 
reg clk;
always #1 clk = ~clk;

reg clr_;
reg set_;

reg VDD;
reg VSS;
reg NOTIFIER;

udp_dff_PWR_scl2mc_cin28hpm_base_svt_c31 u0(out, in, clk, clr_, set_, VDD, VSS, NOTIFIER);

endmodule
```
**lib.v**

```verilog
`timescale 1ns/1ns

`celldefine
primitive udp_dff_PWR_scl2mc_cin28hpm_base_svt_c31 (out, in, clk, clr_, set_, VDD, VSS, NOTIFIER);
output out;
input in, clk, clr_, set_, VDD, VSS, NOTIFIER;
reg out;

table

//in    clk     clr_    set_    VDD     VSS     NOTIFIER    :Qt     :Qt+1
0       r       ?       1       1       0       ?       :   ?   :   0;
1       r       1       ?       1       0       ?       :   ?   :   1;
1       *       1       ?       1       0       ?       :   1   :   1;
0       *       ?       1       1       0       ?       :   0   :   0;
?       f       ?       ?       1       0       ?       :   ?   :   -;
*       b       ?       ?       1       0       ?       :   ?   :   -;
?       ?       ?       0       1       0       ?       :   ?   :   1;
?       b       1       *       1       0       ?       :   1   :   1;
1       x       1       *       1       0       ?       :   1   :   1;
?       ?       0       1       1       0       ?       :   ?   :   0;
?       b       *       1       1       0       ?       :   0   :   0;
0       x       *       1       1       0       ?       :   0   :   0;
?       ?       ?       ?       ?       ?       ?       :   ?   :   x;

endtable //udp_dff_PWR_scl2mc_cin28hpm_base_svt_c31
endprimitive
`endcelldefine

`celldefine
module and(in1, in2, out);
input in1;
input in2;
output out;

assign out = in1 & in2;
endmodule
`endcelldefine
```

here is the command:

```shell
verilator --cc test.v --bbox-unsup -Wno-STMTDLY  --timescale 1ns/1ns  --top-module top -v lib.v
```

wsnyder, 2021-01-22:
As indicated in the error this is in violation of the IEEE spec.  The Verilator documentation indicates how to work around this if you don't care: 

"Error that `timescale is used in some but not all modules. Recommend using --timescale argument, or in front of all modules use:

   `include "timescale.vh"

Then in that file set the timescale.

This is an error due to IEEE specifications, but it may be disabled similar to warnings.  Ignoring this error may result in a module having an unexpected timescale.  [That is use -Wno-TIMESCALEMOD]

closed 2021-01-22 by wsnyder

zhuzhzh, 2021-01-23:
@wsnyder  please check my upd.v. I do have "`timescale 1ns/1ns" in the the file.
My compile command also has --timescale 1ns/1ns.

The issue is when the compiler encounter primitive, then the timescale doesn't work on it.

wsnyder, 2021-01-23:
Ahh, thanks for clarifying.  Fixed to not require timescales on primitives (as they don't need them).

Also made nicer error message that points to primitive rather than the module which is what confused me.

mysoreanoop, 2021-03-08:
@wsnyder 
I'm using the Chipyard framework (which uses an external Verilator, that I've updated to the above mentioned 4.110 version).
It has a verilog simulation flow that is throwing a very similar error:

Am I missing something, or is the --timescale option not seeing everything?

```
%Error-TIMESCALEMOD: /home/anoop/work/git/cmm/sims/verilator/generated-src/redefine.TestHarness.REDEFINEScratchpadConfig/redefine.TestHarness.REDEFINEScratchpadConfig.top.mems.v:457:32: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
  457 |   TS1N28HPCPHVTB1024X8M8SWBASO mem_0_0 (  
      |                                ^~~~~~~
                     /home/anoop/work/git/cmm/sims/verilator/generated-src/redefine.TestHarness.REDEFINEScratchpadConfig/EICG_wrapper.v:3:8: ... Location of module with timescale
    3 | module EICG_wrapper(
      |        ^~~~~~~~~~~~
%Error-TIMESCALEMOD: /home/anoop/work/git/cmm/sims/verilator/generated-src/redefine.TestHarness.REDEFINEScratchpadConfig/redefine.TestHarness.REDEFINEScratchpadConfig.top.mems.v:1417:31: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
 1417 |   TS1N28HPCPHVTB32X50M8SWBASO mem_0_0 (  
      |                               ^~~~~~~
                     /home/anoop/work/git/cmm/sims/verilator/generated-src/redefine.TestHarness.REDEFINEScratchpadConfig/EICG_wrapper.v:3:8: ... Location of module with timescale
    3 | module EICG_wrapper(
      |        ^~~~~~~~~~~~
%Error: Exiting due to 2 error(s)

```
(full log attached as a file)

The Makefile fragment that invokes verilator is below: (there's --timescale option, also tried --timescale-override; but the error remains)
```
# Use --timescale to approximate timescale behavior of pre-4.034
TIMESCALE_OPTS := $(shell verilator --version | perl -lne 'if (/(\d.\d+)/ && $$1 >= 4.034) { print "--timescale 1ns/1ps --timescale-override 1ns/1ps"; }')
VERILATOR_NONCC_OPTS = \
  $(TIMESCALE_OPTS) \
...
```
The modules pointed to by the error are SRAM macros, and they all have timescale declarations in them. However, the ***.mems.v file that instantiates them does not; but that'll be taken care of by the --timescale option, right?

wsnyder, 2021-03-08:
It seems strange that it is giving the error at the instantiation point.  Is this code I could try?   Meantime -Wno-TIMESCALEMOD should work.

mysoreanoop, 2021-03-09:
After using the -Wno-TIMESCALEMOD, the error is bypassed; and reverting to not using the warning-suppression, also works now. Maybe the process was cached, and wasn't up to date. 

Very sorry, if I wasted your time.