verilator/verilator #2764: A way to let Verilator know a DPI import function has dependency on specific reg/wire variables

zenghan2088, 2021-01-23:
Thanks for taking the time to report this.

What would you like added/supported?
Indicate the dependency between reg/wire variables and a DPI import cpp function that refers/changes them.

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
`module top (
    input         clk,
    input  [31:0] din0,
    input  [31:0] din1,
    input  clk_en_0,
    input  clk_en_1,
    output reg [31:0] dout
);

import "DPI-C" context function void dpi_adder();

wire gated_clk_0;
wire gated_clk_1;

clkgate u_clk_gate_0(
  .clk_in(clk),
  .en(clk_en_0),
  .clk_out(gated_clk_0)
);

clkgate u_clk_gate_1(
  .clk_in(gated_clk_0),
  .en(clk_en_1),
  .clk_out(gated_clk_1)
);

reg [31:0] dpi_din0 /* verilator public_flat */;
reg [31:0] dpi_din1 /* verilator public_flat */;
reg [31:0] dpi_dout /* verilator public_flat */;

initial begin
    dpi_din0 = 32'b0;
    dpi_din1 = 32'b0;
    dpi_dout = 32'b0;
end

always @ (posedge gated_clk_0) begin
    dpi_din0 <= din0;
    dpi_din1 <= din1;
end

always @ (posedge gated_clk_1) begin
    $display("VLOG: dpi_din0=%d  dpi_din0=%d", dpi_din0, dpi_din1);
    dpi_adder(); // In c++, it does "dpi_dout = dpi_din0 + dpi_din1" by VPI.
    dout <= dpi_dout;
end`

In above example, $display infers Verilator that dpi_adder() has dependency on dpi_din0 and dpi_din1, so Verilator can order the two always blocks properly. without this $display, the second always is call before the first one in _eval().
"dout <= dpi_out" helps the value change of dpi_out propogate.

Could you add a directive like:
     import "DPI-C" context function void dpi_adder(); /* verilator inputs: dpi_din0, dpi_din1; outputs: dpi_dout */
to help dpi_adder() to be scheduled as expected.

May we assist you in trying to fix this yourself?
glad to

zenghan2088, 2021-01-23:
I just got a better workaround than calling $display. using an empty no inline task.

always @ (posedge gated_clk_1) begin
    dpi_adder_args(dpi_din0, dpi_din1);
    dpi_adder();
    dout <= dpi_dout;
end

task dpi_adder_args(input [31:0] din0, din1);
/* verilator no_inline_task */
// Do nothing
endtask

wsnyder, 2021-01-23:
Glad you found a cleaner workaround.  Your suggestion is good, but relies on some current assumptions in Verilator about when functions need to be sequential.  I'd suggest a slight change:

I just got a better workaround than calling $display. using an empty no inline task.

```
always @ (posedge gated_clk_1) begin
    dpi_adder_args(dpi_din0, dpi_din1);
    dout <= dpi_dout;
end

task dpi_adder_args(input [31:0] din0, din1);
    /* verilator no_inline_task */
    dpi_adder();
endtask
```

Please confirm this works for you (should), and if so I'll add a note to the documentation about this trick.

zenghan2088, 2021-01-24:
Yes, your change works in our real design as well. I insist use mine as it might be more suitable for our verilog.

Thanks for your quick response!

closed 2021-01-25 by wsnyder