verilator/verilator #2765: generate for Can't find definition  in dotted task/function

KGHVerilator, 2021-01-24:
I am creating an instance of an interface inside a for generate loop. I cannot call the functions:

Here is the interface

interface ShiftStageIf #(parameter integer DATA_WIDTH, parameter integer META_WIDTH, parameter integer STAGE);
    typedef struct packed  {
        logic signed[2*DATA_WIDTH-1:0] data;
        logic valid;
        logic last;
        logic [META_WIDTH-1:0] meta;
    } data_t;
    data_t d;

    function automatic void init(input logic signed[DATA_WIDTH-1:0] data_in, input logic valid_in, input logic last_in, input logic [META_WIDTH-1:0] meta_in);
    begin
        d.data = 0;
        d.data[$high(d.data): $size(data_in)] = data_in;
        d.valid = valid_in;
        d.last = last_in;
        d.meta = meta_in;
    end
    endfunction

    function automatic data_t doShift(input logic stall, input logic shift);
        data_t retval;
        if(stall == 1)
            return d;

        retval = d;
        if(shift == 0)
        begin
            return retval;
        end

        retval.data = d.data[$high(d.data)] ? -1 : 0;
        retval.data[$high(d.data)-1: STAGE] = d.data[$high(d.data)-STAGE-1:0];
        return retval;
    endfunction;
endinterface

Inside the code that is failing:

ShiftStageIf #(.DATA_WIDTH(INPUT_WIDTH), .META_WIDTH(METADATA_WIDTH), .STAGE(0)) shiftArray[$high(shift_amount_in)+1:0]();

always_comb
    shiftArray[SHIFT_WIDTH].init(data_in, valid_in, last_in, meta_in);

generate for(genvar i = $high(shift_amount_in); i >= 0; --i)
begin
    ShiftStageIf #(.DATA_WIDTH($size(data_in)), .META_WIDTH($size(meta_in)), .STAGE(i)) p1();

    always_comb
        p1.d = shiftArray[i+1].d;

    always_ff @(posedge sample_clk_in)
        shiftArray[i].d <= p1.doShift(stall, shift_d1.data[i]);
end
endgenerate

verilator --version
Verilator 4.106 2020-12-02 rev UNKNOWN_REV

%Error: /home/hamilton/workspace/d4_decimator_top/logic/src/signal_selection/synth/shift_clip_scale_round_single_sv.sv:83:31: Can't find definition of 'p1' in dotted task/function: 'p1.doShift'
   83 |         shiftArray[i].d <= p1.doShift(stall, shift_d1.data[i]);
      |                               ^~~~~~~
       shift_clip_scale_round_single_sv.sv:9:1: ... note: In file included from shift_clip_scale_round_single_sv.sv
        ... Known scopes under 'doShift': clipRounder, scale_d1, shiftArray[0], shiftArray[1], shiftArray[2], shiftArray[3], shiftArray[4], shiftArray[5], shift_d1
        ... Known scopes under 'doShift': clipRounder, scale_d1, shiftArray[0], shiftArray[1], shiftArray[2], shiftArray[3], shiftArray[4], shiftArray[5], shift_d1
        ... Known scopes under 'doShift': clipRounder, scale_d1, shiftArray[0], shiftArray[1], shiftArray[2], shiftArray[3], shiftArray[4], shiftArray[5], shift_d1
        ... Known scopes under 'doShift': clipRounder, scale_d1, shiftArray[0], shiftArray[1], shiftArray[2], shiftArray[3], shiftArray[4], shiftArray[5], shift_d1
        ... Known scopes under 'doShift': clipRounder, scale_d1, shiftArray[0], shiftArray[1], shiftArray[2], shiftArray[3], shiftArray[4], shiftArray[5], shift_d1
%Error: Exiting due to 1 error(s)

KGHVerilator, 2021-01-24:
Note, this works with a commercial simulator I am using

wsnyder, 2021-01-25:
Certainly seems wrong, could you please turn this into a standalone file that shows the issue and is as simple as possible and will take a look, thanks.

KGHVerilator, 2021-01-25:
interface ShiftIf #(parameter integer WIDTH, parameter integer STAGE);
    typedef struct packed {
        logic [WIDTH-1:0]data;
        logic valid;
    } reg_t;

    reg_t data;

    function automatic reg_t update(input reg_t noChange, input logic enable);
        return enable == 1 ? data : noChange;
    endfunction
endinterface

module automatic gen_loop_if
#(
    parameter integer WIDTH = 8,
    parameter integer SHIFT_WIDTH = 2
)
(
    input  logic logic_clk_in,
    input  logic sync_rst_in,
    input  logic [WIDTH-1:0]dataIn,
    input  logic validIn,
    input  logic enableIn,
    output logic [WIDTH-1:0]dataOut,
    output logic validOut
);

ShiftIf #(WIDTH, 0) shift[SHIFT_WIDTH:0]();
ShiftIf #(WIDTH, 1) shiftTmp1();
ShiftIf #(WIDTH, 2) shiftTmp2();

always_comb
begin
    shift[SHIFT_WIDTH].data.data = dataIn;
    shift[SHIFT_WIDTH].data.valid= validIn;
end

//`define THIS_CODE_FAILS 1
`ifdef THIS_CODE_FAILS

generate for(genvar i = 0; i < SHIFT_WIDTH; ++i)
begin
    ShiftIf #(WIDTH, i) newValue;
    always_comb
        newValue.data = shift[i+1].data;

    always_ff @(posedge logic_clk_in)
        shift[i].data = newValue.update(shift[i].data, enableIn);

end
endgenerate

`else

always_comb
begin
    shiftTmp1.data = shift[1].data;
    shiftTmp2.data = shift[2].data;
end
always_ff @(posedge logic_clk_in)
begin
    shift[0].data = shiftTmp1.update(shift[0].data, enableIn);
    shift[1].data = shiftTmp2.update(shift[1].data, enableIn);
end
`endif

assign dataOut = shift[0].data.data;
assign validOut = shift[0].data.valid;

endmodule

KGHVerilator, 2021-01-25:
Uncomment the line (39):
//`define THIS_CODE_FAILS 1

and it will fail to verilate. Commented out it verilates fine

wsnyder, 2021-02-21:
Forgot to update this earlier, sorry.  Agree this is a problem it's related to how instance arrays are elaborated in two stages.  Using this helps but still hits another related bug (perhaps same one).  Needs more debug attention.

```
  for(genvar i = 0; i < SHIFT_WIDTH; ++i) begin : genw
      ShiftIf #(WIDTH, i) newValue;
      always_comb
        genw[i].newValue.data = shift[i+1].data;
      always_ff @(posedge logic_clk_in)
        shift[i].data = genw[i].newValue.update(shift[i].data, enableIn);
```

Kreijstal, 2024-06-30:
I got caught by this.

how to reproduce.
```
git clone --branch verilator_bug https://github.com/Kreijstal/I2C.git
cd I2C
verilator -y rtl rtl/i2c_master_tb -main --timing --binary --Wno-style --trace-fst 
```
if you try the command 
```
iverilog -o sim_output -y rtl rtl/i2c_master_tb.v  && ./sim_output
```

You will see with iverilog it works.

wsnyder, 2025-11-06:
Simpler test case in #6656.

solomatnikov, 2025-11-06:
@wsnyder any chance this will be fixed soon given that you have simple test case in #6656 ?

wsnyder, 2026-03-30:
Example passes in master which has many recent interface improvements.  Adding this test to regression.