verilator/verilator #2769: [suggestion]allow the mix of ANSI and Non-ANSI module declaration

zhuzhzh, 2021-01-26:
I don't know if it's reasonable. Actually I think users should follow SV LRM.
but many designers writes the non-stardard code. and all the tools compromises.

Here is the coding.

```verilog

module top(input clk, output [31:0] cnt);

reg [31:0] cnt;

always @(posedge clk) begin
    cnt <= cnt + 1;
end

endmodule
```

compile it with VCS,  Xcelium, or check with Spyglass. 
All tools will give out the warning.

Could Verilator give out the warning instead of error as well?

wsnyder, 2021-01-26:
As IEEE says this is an error, VCS says it will "soon" be an error, and this has been illegal since 2001 I think the code should be fixed.

However I would accept a pull request that converts this specific error into a named error e.g. MULDECLANSI or something. Then you will then be able to suppress it or make it into a warning.

wsnyder, 2021-02-21:
See above please.

closed 2021-02-21 by wsnyder