verilator/verilator #2771: Too many preprocessor tokens on a line without any `define

rachitnigam, 2021-01-27:
The attached file is generated by a compiler. The original file is >200,000 lines of system verilog in the same `module`. I've whittled it down to a single assignment that fails.

### Reproduction

Test file: [out.txt](https://github.com/verilator/verilator/files/5882037/out.txt)

```
verilator --lint-only out.txt
```

**Expected**: The file is parsed correctly and reports warnings.
**Actual**: Verilator says:
```
%Error: out.sv:7:57629: Too many preprocessor tokens on a line (>20000); perhaps recursive `define
%Error: out.sv:7:57628: syntax error, unexpected $end, expecting TYPE-IDENTIFIER
%Error: Cannot continue
```

**Note**: There are NO defines in the input file.

**Verilator version**: Verilator 4.108 2021-01-10 rev v4.106-158-g484b76e5b

### Yosys test

I also tried to parse the file with `yosys` which works correctly:
```
yosys -p 'read_verilog -sv out.sv
```

### Workarounds

Since this is generated code, I'm also willing to try out workarounds (for example, changing the `assign` in some manner).

**Would you be willing to try to fix Verilator yourself with assistance?**
Depending on the amount of work represented.

wsnyder, 2021-01-27:
The error means basically your line is too long.  Recursive defines is one way to get that.  If you can't/won't make the line shorter, then recompile verilator with this:

```
diff --git a/src/V3PreProc.h b/src/V3PreProc.h
index a51dab1bf..af86b54a7 100644
--- a/src/V3PreProc.h
+++ b/src/V3PreProc.h
@@ -45,7 +45,7 @@ public:
     // CONSTANTS
     enum MiscConsts {
         DEFINE_RECURSION_LEVEL_MAX = 1000,  // How many `def substitutions before an error
-        LINE_TOKEN_MAX = 20000,  // How many tokens on a line before an error
+        LINE_TOKEN_MAX = 50000,  // How many tokens on a line before an error
         INCLUDE_DEPTH_MAX = 500,  // How many `includes deep before an error
         // Streams deep (sometimes `def deep) before an error.
         // Set more than DEFINE_RECURSION_LEVEL_MAX or INCLUDE_DEPTH_MAX.

```

If that works for you I can commit it.

wsnyder, 2021-02-13:
Assuming the suggestion solved your issue, and/or you fixed your generator.  if you want to mainline the suggested patch please file a pull request, or reopen here.

closed 2021-02-13 by wsnyder

rachitnigam, 2021-02-13:
Ah sorry, I forgot to respond to this. We're going to attempt to reduce the size of the generated assignments to start off. It can't be good for fan-out reasons anyways.

Thanks again for the quick help!

fidesseguisa, 2023-11-20:
Can we possibly increase LINE_TOKEN_MAX to 500,000? We are currently working on a Proof-of-concept using Verilator and our RTL is using a lot of MACROS (one file is around 70K Loc). Updating the RTL is currently not an option for us. Thank you very much in advance.

wsnyder, 2023-11-20:
That's possible, but this is the number of tokens on a single line.  Does it really make lines that long?

The reason for this limitation is to prevent recursive or other hang cases, and 500K might mean a reasonably long pause before an error in some cases.  Can you see what number makes it work, perhaps we set it to the next 100K?

wsnyder, 2023-11-20:
Actually, how about submit a pull request to make this setting an option? (-fpreproc-token-limit)

fidesseguisa, 2023-11-22:
> That's possible, but this is the number of tokens on a single line. Does it really make lines that long?
>
> The reason for this limitation is to prevent recursive or other hang cases, and 500K might [...]

In fact I did, the range is between 400k to 500k.

fidesseguisa, 2023-11-22:
> Actually, how about submit a pull request to make this setting an option? (-fpreproc-token-limit)

Sure, I think this is the better approach. Thank you very much.