verilator/verilator #2772: `vlcovgen` is not executed with python

SamuelRiedel, 2021-01-28:
It seems `src/vlcovgen` has recently been converted from Perl to Python (https://github.com/verilator/verilator/commit/93476a737282f11fea39b6c40ec1b8b2f0628f8b). However, the script is still called with Perl in the build process:
https://github.com/verilator/verilator/blob/05bb24df6c20c76363887094bf68fb6e03c03780/src/Makefile_obj.in#L318

This prevents specifying the Python version in the configure step. Instead, the `vlcovgen` script is called with Perl. It seems to work, but the script's shebang and the system environment will determine the Python version.

``` sh
make[2]: Entering directory `/tools/verilator/src/obj_dbg'
/usr/bin/perl ../vlcovgen --srcdir ..
```

wsnyder, 2021-01-28:
Thanks for the debug & fix.

closed 2021-01-28 by wsnyder

SamuelRiedel, 2021-01-28:
:+1: Thanks for the quick response.

PR #2773 (2021-01-28): Execute vlcovgen with python3.
This fixes #2772.

The `src/vlcovgen` script was converted from Perl to Python, but the build system still executes the script with Perl. This PR fixes this.

This is my first contribution to Verilator. I certify that I comply with the developer's certificate. I also added my name to the contributor's list. Please let me know if you need anything else.