verilator/verilator #2774: array of bits cannot be compiled

mytbk, 2021-01-29:
The following SystemVerilog code can be translated to C++ without any warning or error, but fails to compile.

```
class MyClass;
	bit [15:0] data[4];
endclass
```
error:

```
$ ~/verilator/bin/verilator --cc --build reg.sv 
make: Entering directory '/home/irudog/programs/verilog/sv/obj_dir'
/usr/bin/perl /home/irudog/verilator/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vreg.cpp Vreg___024unit.cpp Vreg___024unit__03a__03aMyClass__Vclpkg.cpp Vreg__Slow.cpp Vreg___024unit__Slow.cpp Vreg___024unit__03a__03aMyClass__Vclpkg__Slow.cpp Vreg__Syms.cpp > Vreg__ALL.cpp
ccache g++  -I.  -MMD -I/home/irudog/verilator/share/verilator/include -I/home/irudog/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o Vreg__ALL.o Vreg__ALL.cpp
In file included from Vreg__ALL.cpp:5:
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp: In member function ‘std::string Vreg___024unit__03a__03aMyClass::to_string_middle() const’:
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp:48:46: error: no matching function for call to ‘VL_TO_STRING(const SData [4])’
   48 |     out += "data:" + VL_TO_STRING(__PVT__data);
      |                                              ^
In file included from Vreg.h:11,
                 from Vreg.cpp:5,
                 from Vreg__ALL.cpp:3:
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:40:20: note: candidate: ‘std::string VL_TO_STRING(CData)’ (near match)
   40 | extern std::string VL_TO_STRING(CData lhs);
      |                    ^~~~~~~~~~~~
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:40:20: note:   conversion of argument 1 would be ill-formed:
In file included from Vreg__ALL.cpp:5:
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp:48:35: error: invalid conversion from ‘const SData*’ {aka ‘const short unsigned int*’} to ‘CData’ {aka ‘unsigned char’} [-fpermissive]
   48 |     out += "data:" + VL_TO_STRING(__PVT__data);
      |                                   ^~~~~~~~~~~
      |                                   |
      |                                   const SData* {aka const short unsigned int*}
In file included from Vreg.h:11,
                 from Vreg.cpp:5,
                 from Vreg__ALL.cpp:3:
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:41:20: note: candidate: ‘std::string VL_TO_STRING(SData)’ (near match)
   41 | extern std::string VL_TO_STRING(SData lhs);
      |                    ^~~~~~~~~~~~
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:41:20: note:   conversion of argument 1 would be ill-formed:
In file included from Vreg__ALL.cpp:5:
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp:48:35: error: invalid conversion from ‘const SData*’ {aka ‘const short unsigned int*’} to ‘SData’ {aka ‘short unsigned int’} [-fpermissive]
   48 |     out += "data:" + VL_TO_STRING(__PVT__data);
      |                                   ^~~~~~~~~~~
      |                                   |
      |                                   const SData* {aka const short unsigned int*}
In file included from Vreg.h:11,
                 from Vreg.cpp:5,
                 from Vreg__ALL.cpp:3:
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:42:20: note: candidate: ‘std::string VL_TO_STRING(IData)’ (near match)
   42 | extern std::string VL_TO_STRING(IData lhs);
      |                    ^~~~~~~~~~~~
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:42:20: note:   conversion of argument 1 would be ill-formed:
In file included from Vreg__ALL.cpp:5:
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp:48:35: error: invalid conversion from ‘const SData*’ {aka ‘const short unsigned int*’} to ‘IData’ {aka ‘unsigned int’} [-fpermissive]
   48 |     out += "data:" + VL_TO_STRING(__PVT__data);
      |                                   ^~~~~~~~~~~
      |                                   |
      |                                   const SData* {aka const short unsigned int*}
In file included from Vreg.h:11,
                 from Vreg.cpp:5,
                 from Vreg__ALL.cpp:3:
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:43:20: note: candidate: ‘std::string VL_TO_STRING(QData)’ (near match)
   43 | extern std::string VL_TO_STRING(QData lhs);
      |                    ^~~~~~~~~~~~
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:43:20: note:   conversion of argument 1 would be ill-formed:
In file included from Vreg__ALL.cpp:5:
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp:48:35: error: invalid conversion from ‘const SData*’ {aka ‘const short unsigned int*’} to ‘QData’ {aka ‘long unsigned int’} [-fpermissive]
   48 |     out += "data:" + VL_TO_STRING(__PVT__data);
      |                                   ^~~~~~~~~~~
      |                                   |
[... 28 lines cut ...]
      |                                              ^
In file included from Vreg.h:11,
                 from Vreg.cpp:5,
                 from Vreg__ALL.cpp:3:
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:765:13: note: candidate: ‘template<class T_Key, class T_Value> std::string VL_TO_STRING(const VlAssocArray<T_Key, T_Value>&)’
  765 | std::string VL_TO_STRING(const VlAssocArray<T_Key, T_Value>& obj) {
      |             ^~~~~~~~~~~~
/home/irudog/verilator/share/verilator/include/verilated_heavy.h:765:13: note:   template argument deduction/substitution failed:
In file included from Vreg__ALL.cpp:5:
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp:48:46: note:   mismatched types ‘const VlAssocArray<T_Key, T_Value>’ and ‘const SData [4]’ {aka ‘const short unsigned int [4]’}
   48 |     out += "data:" + VL_TO_STRING(__PVT__data);
      |                                              ^
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp:32:13: note: candidate: ‘std::string VL_TO_STRING(const std::shared_ptr<Vreg___024unit__03a__03aMyClass>&)’
   32 | std::string VL_TO_STRING(const VlClassRef<Vreg___024unit__03a__03aMyClass>& obj) {
      |             ^~~~~~~~~~~~
Vreg___024unit__03a__03aMyClass__Vclpkg.cpp:32:77: note:   no known conversion for argument 1 from ‘const SData [4]’ {aka ‘const short unsigned int [4]’} to ‘const std::shared_ptr<Vreg___024unit__03a__03aMyClass>&’
   32 | std::string VL_TO_STRING(const VlClassRef<Vreg___024unit__03a__03aMyClass>& obj) {
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
make: *** [/home/irudog/verilator/share/verilator/include/verilated.mk:235: Vreg__ALL.o] Error 1
make: Leaving directory '/home/irudog/programs/verilog/sv/obj_dir'
%Error: make -C obj_dir -f Vreg.mk exited with 2
%Error: Command Failed /home/irudog/verilator/bin/verilator_bin --cc --build reg.sv
```

I've tried both version 4.108 and git revision 4.109 devel rev v4.108-27-g483516ce.

mytbk, 2021-02-01:
I just find out that using ``bit [3:0][15:0] data`` instead of ``bit [15:0] data[4]`` can make the source built.
Also, I find that ``bit data[4]`` cannot be built either.

wsnyder, 2021-02-01:
Yes, all unpacked arrays are broken, and using a packed array is a good workaround.  It's a bit complicated to fix as hit other problems, will get to it.

wsnyder, 2021-03-13:
This bug is related to handling the stringification of packed arrays.  To handle this, all arrays and wide types must be known as C++ types rather than basic C arrays so the proper conversion functions can be called.

As part of this will soon be committing:
* All old "WData signal[WORDS]" declarations which were used for signals > 64 bits will now be "VlWide<WORDS> signal"
* All old "type signal[DEPTH]" unpacked declarations will now be "VlUnpacked<type, DEPTH> signal"
* Compile time went down for this change 10% which I don't understand.
* Runtime went up 2% which is acceptable, later cleanups may get this back.

Previously VlWide and VlUnpacked were only used when certain associative arrays and other constructs needed them, they will now always be used.

Appropriate conversions are in place, but as these changes propagate to the header of a Verilated model, it's possible this will upset some user wrappers.

closed 2021-03-13 by wsnyder