verilator/verilator #2778: VPI: simulation time callbacks should be one-time

marlonjames, 2021-02-04:
Verilator version: master fadc677

Currently, for **simulation time callbacks** (see IEEE 1800-2017 38.36.2) other than `cbAfterDelay`, Verilator ignores the _cb_data_p->time_ field, and they are registered as repeated callbacks rather than one-time callbacks.

After some investigation, I've found that iverilog and aldec simulators both register these as one-time callbacks (e.g.,  `cbReadWriteSynch`), and use the _time_ field to schedule the callback.

Verilator VPI should match this behavior.

wsnyder, 2021-02-13:
Might you be able to submit a pull request to modify a test_regress test to show correct behavior, make sure it passes on other simulators, and ideally fix this behavior?

marlonjames, 2021-02-16:
Yes, I'll try to find time for this soon.

nickoe, 2021-06-14:
@marlonjames Did you ever get a real chance to look into this issue?

marlonjames, 2021-06-14:
I started on this, but I ran into issues with a test for the callbacks. I had a test working with iverilog but when I ported it to the verilator test framework it failed and I couldn't figure out why.
I haven't had time to resolve it and won't in the near future.

If anybody wants to work on this, you could start with my branch that stripped down the previous callback test at https://github.com/marlonjames/verilator/tree/refactor_cb_test.

I would probably tackle the tests one callback at a time, as the _time_ field means different things for different callbacks, and it makes a combined test complicated.
See https://github.com/steveicarus/iverilog/issues/490 for `cbAtStartOfSimTime` / `cbAtEndOfSimTime `.

nickoe, 2021-06-14:
I will not be the right guy for that, I am just a cocotb user on arch which uses the latest verilator... which does not currently work with cocotb, I guess partially because the support is only considered "experimental" and this very issue. :)

kammoh, 2021-07-21:
Any updates on this? 
`cocotb` does not work with Verilator since 4.108 :(

ktbarrett, 2021-07-25:
Introduced in 1e34ae31d2b09094f96bcdd606daa47676c77e76.

A simple fix that makes cocotb function:
```patch
diff --git a/include/verilated_vpi.cpp b/include/verilated_vpi.cpp
index c354d89f..64099b98 100644
--- a/include/verilated_vpi.cpp
+++ b/include/verilated_vpi.cpp
@@ -603,6 +603,7 @@ public:
             VerilatedVpiCbHolder& ho = *it;
             VL_DEBUG_IF_PLI(VL_DBG_MSGF("- vpi: reason_callback reason=%d id=%" VL_PRI64 "d\n",
                                         reason, ho.id()););
+            ho.invalidate();
             (ho.cb_rtnp())(ho.cb_datap());
             called = true;
             if (was_last) break;
```

Not sure if this is the "right" solution.

kammoh, 2021-07-26:
@ktbarrett I can confirm that the suggested fix works!

wsnyder, 2021-07-26:
@ktbarrett would you be able to try converting this into a pull request?

1. Please add or modify an existing test to show that the case fails without the patch, then passes with it.
2. I can't see where in IEEE it makes clear what events are one-shot versus repeated.  Your patch suggests all are one shot (as you always call invalidate), if that's not the case you instead need an "if (cb == whatevers) ho.invalidate();".  That might be right but I'm not sure, so to prove this out we need the tests to also run against iverilog (or another simulator) to make sure there's agreement.

ktbarrett, 2021-07-26:
> I can't see where in IEEE it makes clear what events are one-shot versus repeated. 

I was also unable to find anything.

> Your patch suggests all are one shot

This is what cocotb assumes because it's what all other simulator's VPIs assume, at least for timing callbacks.

> if that's not the case you instead need an "if (cb == whatevers) ho.invalidate();"

Certainly there are some callbacks that wouldn't really make sense to only be one-shot. It's hard to say what other simulators think because most of them don't support them :laughing: I'll condition invalidation on timing triggers only and let someone else deal with the rest.

marlonjames, 2021-08-11:
There are several things to consider, and several ambiguities within the standard.
There is absolute vs relative time as mentioned earlier ( steveicarus/iverilog#490 ).

Also, because there are several timed callbacks, `cbNextDeadline()` will have to take them into account, and there will probably need to be a new way to get all the callback types that are pending for a timestep. Most of these callbacks are called through `callCbs(<cb reason>)`, with `cbAfterDelay` callbacks called using the special `callTimedCbs`.
The user writing the main loop will need to know if callbacks are pending for each reason in each timestep.

ktbarrett, 2021-09-24:
@marlonjames I wouldn't let the best solution get in the way of a solution.

alexforencich, 2021-11-09:
Any movement on this? This has been open for 9 months now, and verilator is unusable without the fix for this.

wsnyder, 2021-11-09:
@alexforencich waiting for a pull request, perhaps you'd attempt one?

alexforencich, 2021-11-09:
So if I just submit https://github.com/verilator/verilator/issues/2778#issuecomment-886149648 as a pull request, would that be sufficient?

wsnyder, 2021-11-09:
Need tests too, and ones that pass on other simulators i.e. NC and/or VCS.

alexforencich, 2021-11-09:
I don't have access to NC, VCS, etc. So I will have to defer to someone else in that case.

alexforencich, 2021-11-09:
Alternatively, how about we bisect and revert a whole commit without any extra tests, since this is a regression? Then the commit in question can be fixed and reapplied later once the necessary tests can be put together.

wsnyder, 2021-11-09:
Or Icarus, but not sure if it passes the tests even without this test....

wsnyder, 2021-11-09:
There's too many changes to backtrack.  I guess if you make a pull with the fix and a Verilator-passing test without another simulator I'll take it.

alexforencich, 2021-11-09:
It's not that many commits to check, just from 4.106 (where it works, released Dec. 2) and 4.108 (where it is broken, released Jan 10).

ktbarrett, 2021-11-09:
@alexforencich I ran a bisect and found the offending commit. https://github.com/verilator/verilator/issues/2778#issuecomment-886149648

alexforencich, 2021-11-09:
Oh, I totally didn't see that line. Github really minimizes the reference to the commit. That's quite a large commit though, not sure if outright reverting it would cause other problems at this point.

alexforencich, 2021-11-10:
Well, apparently @ktbarrett's fix is not sufficient; `t_vpi_cbs_called` fails with that change.  Unfortunately, I do not know enough about verilator internals or VPI callbacks to take a proper pass at fixing this regression.

alexforencich, 2021-11-10:
I spent a bit more time poking around and have conditioned the fix on only the time-based callbacks, like so:

            switch (reason) {
            case cbAtStartOfSimTime:  // FALLTHRU
            case cbNBASynch:  // FALLTHRU
            case cbReadWriteSynch:  // FALLTHRU
            case cbAtEndOfSimTime:  // FALLTHRU
            case cbReadOnlySynch:  // FALLTHRU
            case cbNextSimTime:
                ho.invalidate();  // Simulation time callbacks are one-shot
                break;
            }

But the test case `t_vpi_cbs_called` still fails, presumably because it assumes that all of the callbacks are not one-shot callbacks.  So, presumably the test will also need to be changed, which now puts us right back where we started: what, exactly, is the correct behavior in this case, if the spec doesn't shed any light on this?

marlonjames, 2021-11-10:
These should be one-time callbacks.
When registering for a callback, the VPI application can provide a time delay before the callback is called, allowing scheduling callbacks for the future. Verilator does not take the time into account at the moment.

If the callbacks were tested with a time of 0, it may pass *some* other simulators, and be a partial solution to the issue.

A big problem is that the VPI implementations are not consistent across simulators. For instance, a time of 0 will cause an assert in iverilog for `cbAtStartOfSimTime` if the current simulation time is greater than 0.

alexforencich, 2021-11-10:
TBH, I have never directly interacted with the VPI, so I'm not sure what callbacks are available and exactly how they're supposed to behave.  

It seems like the spec doesn't make it clear if these are supposed to be one-shot or not: https://github.com/verilator/verilator/issues/2778#issuecomment-887102793 .  

I suppose one issue with Verilator is that since it doesn't really have a scheduler, it can't really delay anything based on time, only based on various events.  But I suppose this is not necessarily a problem so long as cocotb can work around it.

According to the spec, "Attempting to place a cbAtStartOfSimTime callback with a delay of zero when simulation has progressed into a time slice and the application is not currently within a cbAtStartOfSimTime callback." so does that mean iverilog is correct with causing an assert failure, since you shouldn't call it that way in the first place?

marlonjames, 2021-11-10:
It depends on when in the time step the VPI application attempts to register it.
More precisely, if you attempt to register for a `cbAtStartOfSimTime` callback with a time that is less than the current time, it will assert. This is because iverilog treats the time as an absolute time, not a "delay" relative to the current time.
If you look at https://github.com/steveicarus/iverilog/issues/490, he references one of the main contributors of the PLI/VPI section of the standard as reason for why iverilog does it that way.
The language in the standard is not clear, thus the differing implementations and arguments for which is "correct".

alexforencich, 2021-11-10:
Why in the *&(#$ would ANY API allow you to specify an absolute time for an event-driven simulator....if you want an absolute time for some reason, read the current sim time, subtract that from the absolute time you want to get a relative time, and use that.  It makes everything super simple - all positive values are OK, and possibly zero depending on the call.  Anything negative is in the "past" and hence is an error.

But I suppose that does raise the question, what is the difference between cbAfterDelay and cbAtStartOfSimTime?  If they are the same, why not just ignore cbAtStartOfSimTime and always use cbAfterDelay?

markusdd, 2022-01-16:
I have also participated in debugging and testing for cocotb/verilator interfaces in the past, is there anything I can try to do for you guys?

Would like to update to a newer verilator and recetn cocotb as well but this is kind of a long standing showstopper right now.

marlonjames, 2022-01-16:
@markusdd From an earlier comment of mine:
> Also, because there are several timed callbacks, `cbNextDeadline()` will have to take them into account, and there will probably need to be a new way to get all the callback types that are pending [...]
The user writing the main loop will need to know if callbacks are pending for each reason in each timestep.

Do you have any suggestions on what the updated interface should look like?

kammoh, 2022-09-09:
Any updates or are we still interpreting the standard?

marlonjames, 2022-09-09:
I have an in-progress branch with a re-worked test that passes Icarus.
The remaining step is to make Verilator match the VPI spec, and provide a new interface to the top-level simulation loop that provides information about all callback types that are scheduled for the time-step.

kammoh, 2022-09-09:
> I have an in-progress branch with a re-worked test that passes Icarus.

Awesome!
Is this a patch to Icarus or Verilator?
In case the branch is publicly available, I can try it on a few designs and platforms and report back.
I'd also be happy to help with the verilator changes, as long as I know the consensus on the "correct" approach around the issue.

yacremann, 2022-10-26:
Hi
Thanks for all your effort on this, this is very important! If there is something to test with Verilator in combination with Cocotb we have a lot of designs in SystemVerilog where we also have testbenches. For us the combination of Cocotb with Verilator is the only way we can simulate designs in SystemVerilog using free (as in freedom) tools.
So if I can help with testing something using our designs please let me know!

markusdd, 2022-11-13:
Hi everyone, I'm also happy to offer my help to benchmark this against my test suites.

Do we have any rough ETA when something might be available? Being stuck on a 2 year old version slowly evolves into an actual issue. I'm ready to invest time into testing potential fixes or whatever but I fear I'm not of any real help when it comes to VPI itself, I'm just not familiar enough with it.

wsnyder, 2022-11-13:
As was requested back in Feb 21, we're still waiting on someone that knows the VPI to provide tests that show correct behavior (which passes on old Verilator, fails on new Verilator, and passes on other simulators.)  No ETA until we have that.

ProceduralInterface, 2022-11-18:
Its a shame the standard is not clear, I have a reference implementation from circa 2005 standards committee which provides some clarity. It doesn't appear things have changed since, except `vpi_free_object` deprecated for `vpi_release_handle` in 2009.

One-shot (includes all time events):
- `cbReadOnlySynch`
- `cbReadWriteSynch`
- `cbAtStartOfSimTime`
- `cbAtEndOfSimTime`
- `cbAfterDelay`
- `cbNextSimTime`
- `cbEndOfCompile`
- `cbStartOfSimulation`
- `cbEndOfSimulation`
- `cbStartOfSave`
- `cbEndOfSave`
- `cbStartOfRestart`
- `cbEndOfRestart`
- `cbNBASync`

ktbarrett, 2022-11-18:
@ProceduralInterface Is it possible you could share that reference implementation?

markusdd, 2022-11-18:
That would certainly appreciated.
Meanwhile, @ktbarrett do you have an affected testcase on hand?
Otherwise I could procure a repository with a design and coco testbench that works at 106 but does not with later versions.

would that be of any help? Because I have no idea obout the VPI but I'd like to help drive this forward any way I can.

dramoz, 2022-11-18:
Hi, in the meantime if someone needs the latest Verilator with [CoCoTB](https://docs.cocotb.org/en/stable/index.html), I am maintaining an up-to-date Verialtor fork with [current proposed workaround](https://github.com/verilator/verilator/issues/2778#issuecomment-886149648). The patch is applied to the **latest tag** in a separate **branch: _vtag_-vpi-callback-patch**. The master is untouched (except for the  `README.rst` file)

https://github.com/Eideticom/verilator
https://github.com/Eideticom/verilator/tree/v5.002-vpi-callback-patch

ProceduralInterface, 2022-11-18:
> @ProceduralInterface Is it possible you could share that reference implementation?
If only it were that easy - you'd run the big 3 out of business 😉!

I think the proposed fix is horrible - easy to improve, add a reason check for one shot CBs. i.e. we should invalidate only for (`cbEndOfCompile`,`cbStartOfSimulation`,`cbEndOfSimulation`,`cbStartOfSave`,`cbEndOfSave,`cbStartOfRestart`,`cbEndOfRestart`)

And stop using `callCbs` for any of these: `cbReadOnlySynch`,`cbReadWriteSynch`,`cbAtStartOfSimTime`,`cbAtEndOfSimTime`,`cbAfterDelay`,`cbNextSimTime`,`cbNBASync`
which should roughly copy the current implementation of `cbAfterDelay` (except other use absolute time not relative)

markusdd, 2022-11-18:
I have just built and checked the patched version by @dramoz and ran it against some of my simple and more compley tests and it works well. (base: verilator 5.002)

So if someone can pick up the proposal by @ProceduralInterface I am more than happy to test that as well.

kammoh, 2022-11-21:
> > @ProceduralInterface Is it possible you could share that reference implementation?
> > If only it were that easy - you'd run the big 3 out of business 😉!
> 
> I think the proposed fix is 'sub-optimal' - easy to improve, add a reason check for one shot CBs. i.e. we should invalidate only for (`cbEndOfCompile`,`cbStartOfSimulation`,`cbEndOfSimulation`,`cbStartOfSave`,`cbEndOfSave`,`cbStartOfRestart`,`cbEndOfRestart`)

Would this be the only required change?
https://github.com/verilator/verilator/compare/master...kammoh:verilator:one_time_callbacks

ProceduralInterface, 2022-11-21:
Definitely an improvement but as markusdd mentions we need a new interface which resolves the time based interrupts in time and event type priority order closer to the standard, using `callCbs` for the time scheduled callbacks is wrong but this is a better fix than the prior suggestion. I will have a go, but it's a rather large job if we need to design and test a more complex ordered event heap.

yacremann, 2023-01-11:
> 
This is great, thanks a lot! I will keep testing it using our designs.

yacremann, 2023-01-11:
> Hi, in the meantime if someone needs the latest Verilator with [CoCoTB](https://docs.cocotb.org/en/stable/index.html), I am maintaining an up-to-date Verialtor fork with [current proposed [...]

Thanks a lot! This is great and really important for us. We will keep testing it with our designs.

marlonjames, 2023-01-12:
If anybody wants to implement the Verilator VPI support for this, I have a branch with reworked tests: https://github.com/marlonjames/verilator/tree/onetime_cbs

- Reduced current test to just `cbValueChange` for repetitive callback test.
- Added new test for one-time callbacks.

Both tests pass using iverilog. The repetitive callback test passes with Verilator.

Verilator implementation needs:
- Registering scheduled one-time callbacks with a non-NULL time, and track callback time similar to `cbAfterDelay`
- Some `cbNextDeadline()` equivalent for non-`cbAfterDelay` callbacks
- Modify `callCbs()` to call one-time callbacks registered for the current time step, but not future scheduled callbacks
- *Optional: Some function to get list of callback reasons that have registered callbacks for the current time step.*

wsnyder, 2023-01-21:
@marlonjames thanks for updating the test, that's a great help.  The test core dumps for me on Icarus, but I ignored that and was able to debug the test on Verilator assuming it was known-good.  The only functional change of note I made to the test was it was missing a call to `VerilatedVpi::callCbs(cbAtStartOfSimTime)`.

I committed the fixes so the test now passes, and I also added the debug dump.  Please try this on Cocotb - if there's any issues still I'd appreciate if you could update the test to show the issue.  I'll leave this issue open until I hear back.

wsnyder, 2023-01-22:
Think we are ready for a release once this can be CocoTB validated (or will go without if takes too long!)

markusdd, 2023-01-22:
Hi, as I offered to help with testing I could build verilator locally and run it against my testcases that were failing without this fix.

Can you point me to the commit I should build against?

wsnyder, 2023-01-22:
You can use master please.

markusdd, 2023-01-22:
Guys, that looks fantastic. Thanky so SO much! All test suites are passing, one example in screenshot below including verilator version. CocoTB version was 1.71.

[image]
[image]

wsnyder, 2023-01-22:
Great, good community effort here!  Closing.

closed 2023-01-22 by wsnyder