verilator/verilator #2781: How to add signals to the VCD dump generated from code external to Verilator

jotego, 2021-02-06:
I am designing a Verilog module, and I have a C model for that module. I can simulate the module in Verilator and produce a VCD dump. I want to add to this VCD the signals from my model -which runs in the same C++ program- to compare the two outputs in the waveform viewer.

I have a VerilatedVcdC object, which the Verilator object uses to dump. Can I add more signals to that VerilatedVcdC object? I have tried doing _spTrace()->declBus_ before starting the simulation, and then _spTrace()->chgBus_ after each clock cycle but that seems to break the VerilatedVcdC object as no dump is produced.

Could you give me some hints, please?

wsnyder, 2021-02-06:
There is no currently supported way to directly add your own signal to a C++ VCD dump.  Some alternatives that work today are

1. Use SystemC and you can use SC dumps to get your signal into a VCD.
2. Add whatever signal is of interest to your verilog design as an input.
3. Add whatever signal is of interest to a new verilog design that only has that input and simulate that and your other design in parallel.  For an example with two designs see test_regress/t/t_trace_two_cc.cpp

It's also possible to improve the code to make this supported as it once was, but that will be more effort.

closed 2021-02-06 by wsnyder

jotego, 2021-02-06:
Thank you for your kind answer.