verilator/verilator #2782: false BLOCKINIT

zhuzhzh, 2021-02-08:
$ verilator --version                                  
Verilator 4.103 devel rev v4.102-35-g0c949ceb

I created the small case as the following. It works.

```verilog
module test(input clk);

reg [31:0] dout [15:0];

always @(posedge clk) begin
    integer i;
    for(i=0;i<16;i=i+1) begin
        dout[i] <= 32'd0;
    end 
end

endmodule
```

but in the real case, it reports the BLOCKINIT error.

```verilog

module malu_in_pproc(/*AUTOARG*/
    clk,
   // Outputs
   vpt_malu_data_1d ,vpt_malu_ld_vld,
   // Inputs
   vpt_malu_data, vpt_malu_vld, vpt_malu_opcode,vpt_malu_mode,vpt_malu_vad,vpt_malu_union,fp32_flush_out,vpt_malu_dual,fp16_flush_in,bf16_flush_in
   );

   input          clk;
   input [1023:0] vpt_malu_data;
   input          vpt_malu_vld;
   input [4:0]    vpt_malu_opcode;
   input [4:0]    vpt_malu_mode;
   input [1:0]    vpt_malu_vad;
   input [1:0]    vpt_malu_union; 
   input          vpt_malu_dual;
   input          fp32_flush_out;
   input          fp16_flush_in;
   input          bf16_flush_in;
   
   output[2047+64:0] vpt_malu_data_1d;
   output reg [3:0]    vpt_malu_ld_vld;

    parameter [4:0] MADD    = 5'd0;
    parameter [4:0] MSUB    = 5'd1;
    parameter [4:0] MDM     = 5'd2;
    parameter [4:0] VDM_ROW = 5'd3;
    parameter [4:0] VDM_COL = 5'd4;
    parameter [4:0] VCMPAC  = 5'd5;
    parameter [4:0] LDSMR   = 5'd6;
    parameter [4:0] SVMM    = 5'd7;
    parameter [4:0] VMM     = 5'd8;
    parameter [4:0] VMAC    = 5'd9;
    parameter [4:0] VMAS    = 5'd10;
    parameter [4:0] IMAS    = 5'd11;

    parameter [4:0] MOVVA   = 5'd12;
    parameter [4:0] VLDI_LO = 5'd13;
    parameter [4:0] VLDI_HI = 5'd14;
    parameter [4:0] VSTLA   = 5'd15;
    parameter [4:0] VSTA    = 5'd16;
    parameter [4:0] TVSTA   = 5'd17;
    parameter [4:0] VA2VR   = 5'd18;
    parameter [4:0] VCLR    = 5'd19;
    parameter [4:0] R2VA    = 5'd20;
    parameter [4:0] TVLDA   = 5'd21;
[... 453 lines cut ...]
        if (vpt_malu_vld) begin
            for (i=0;i<16;i=i+1) begin
                case (vpt_malu_opcode)
                    VCLR : begin
                        vpt_malu_bank0[i] <= 32'd0;
                        vpt_malu_bank1[i] <= 32'd0;
                        vpt_malu_bank2[i] <= 32'd0;
                        vpt_malu_bank3[i] <= 32'd0;
                        case (vpt_malu_union)
                            //2'd0 : begin                                                                //comment out this branch statement
                            //    case (vpt_malu_vad[1:0])
                           //         2'd0 : vpt_malu_ld_vld    <= 4'b0001;
                           //        2'd1 : vpt_malu_ld_vld    <= 4'b0010;
                           //         2'd2 : vpt_malu_ld_vld    <= 4'b0100;
                          //          2'd3 : vpt_malu_ld_vld    <= 4'b1000;
                         //           default : vpt_malu_ld_vld <= 4'b0000;
                         //       endcase
                        //    end
```

Actually,  If I comment out any code, the issue will be gone. (it should not be so sensitive).
So I think maybe it's one coner case.

Please help check it.
besides, I tried Xcelium and VCS. both of them work.

wsnyder, 2021-02-13:
I suspect there is too much code in the loop body, which will disable the loop unrolling optimization, this is why it works OK with less code.  Either shorten the amount of code (perhaps using a temp var), or try e.g. "--unroll-stmts 100000".

zhuzhzh, 2021-02-19:
Thanks, @wsnyder .

With "--unroll-stmts 100000", the original complete code can pass now.

closed 2021-02-22 by wsnyder