verilator/verilator #2783: Typedef of parameterized structure in interface

zegervdv, 2021-02-08:
Thanks for taking the time to report this.

What would you like added/supported?

While parameterized structs are not directly supported in Systemverilog, there is a workaround using structs defined in interfaces.
(see https://verificationacademy.com/forums/systemverilog/parameterized-struct-systemverilog-design).

The idea is to define the struct in the interface, using those parameters. The interface instance is then passed to the module, where a local typedef is needed to make the type accessible inside the module.

I am getting the error that that typedef is not supported:
```
$ verilator -Wall --default-language 1800-2017 --cc foo.sv 
%Error-UNSUPPORTED: foo.sv:15:3: Unsupported: SystemVerilog 2005 typedef in this context
   15 |   typedef bar_if.struct_t struct_t;
      |   ^~~~~~~
```

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

Minimal example:
``` systemverilog
interface bar #(
  parameter int unsigned param
) ();
  typedef struct {
    logic valid;
    logic [param-1:0] data;
  } struct_t;
endinterface

module foo #() (
  input logic                 i_clk,
  input logic                 i_rst,
  bar bar_if
);
  typedef bar_if.struct_t struct_t;
endmodule
```

May we assist you in trying to fix this yourself?

wsnyder, 2021-02-17:
Everything with interfaces seems to turn out harder then expected... have this parsing in a branch but there's complications in elaboration.  Will need to get back to it later.

solomatnikov, 2023-06-02:
https://github.com/verilator/verilator/issues/3441#issuecomment-1573063210