verilator/verilator #2784: [verilator@3.920 make test failure] Self-test cases fail to be executed after the Spack is installed.

Tom-python0121, 2021-02-09:
Hello,I meet a question about Self-test cases fail to be executed after the Spack is installed of verilator .
### Steps to reproduce the issue

```console
[root@centos8 ~]#spack install --keep-stage verilator@3.920
[root@centos8 ~]#spack cd verilator
[root@centos8 spack-src]#spack load verilator
[root@centos8 spack-src]#source ../spack-build-out.txt
[root@centos8 spack-src]# cd test_regress/
[root@centos8 test_regress]#make test
……
TYPES    Vt_wire_types    > obj_dir/t_wire_types/vlt_gcc.log
make[1]: Entering directory '/tmp/root/spack-stage/spack-stage-verilator-3.920-74zk5x4hvr2bj2cwyb2btym32hy3gdau/spack-src/test_regress/obj_dir/t_wire_types'
/home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.3.1/perl-5.30.3-2hodatpomwzo53za4danhjcthvxzbava/bin/perl /tmp/root/spack-stage/spack-stage-verilator-3.920-74zk5x4hvr2bj2cwyb2btym32hy3gdau/spack-src/test_regress/../bin/verilator_includer -DVL_INLINE_OPT=inline Vt_wire_types__main.cpp /tmp/root/spack-stage/spack-stage-verilator-3.920-74zk5x4hvr2bj2cwyb2btym32hy3gdau/spack-src/test_regress/../include/verilated.cpp Vt_wire_types.cpp Vt_wire_types__Syms.cpp > Vt_wire_types__ALLboth.cpp
/home/spack/spack/lib/spack/env/gcc/g++  -I.  -MMD -I/tmp/root/spack-stage/spack-stage-verilator-3.920-74zk5x4hvr2bj2cwyb2btym32hy3gdau/spack-src/test_regress/../include -I/tmp/root/spack-stage/spack-stage-verilator-3.920-74zk5x4hvr2bj2cwyb2btym32hy3gdau/spack-src/test_regress/../include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -DVERILATOR=1 -DVL_DEBUG=1 -DVM_PREFIX=Vt_wire_types -DVM_PREFIX_INCLUDE="<Vt_wire_types.h>" -DT_WIRE_TYPES    -c -o Vt_wire_types__ALLboth.o Vt_wire_types__ALLboth.cpp
/home/spack/spack/lib/spack/env/gcc/g++    -g Vt_wire_types__ALLboth.o    -o Vt_wire_types -lm -lstdc++ 2>&1
make[1]: Leaving directory '/tmp/root/spack-stage/spack-stage-verilator-3.920-74zk5x4hvr2bj2cwyb2btym32hy3gdau/spack-src/test_regress/obj_dir/t_wire_types'
vlt/t_wire_types: Run
        obj_dir/t_wire_types/Vt_wire_types    > obj_dir/t_wire_types/vlt_sim.log
*-* All Finished *-*
- t/t_wire_types.v:59: Verilog $finish
vlt/t_wire_types: Test PASSED
==SUMMARY: Left NO-FORKER  Passed 785  Unsup 15  Skipped 35  Failed 73
======================================================================
vlt/t_xml_first: ==================================================
vlt/t_xml_first: Compile
        perl ../bin/verilator --prefix Vt_xml_first --x-assign unique -cc -Mdir obj_dir/t_xml_first -OD --debug-check --comp-limit-members 10 --xml-only --clk clk  -f input.vc t/t_xml_first.v    > obj_dir/t_xml_first/vlt_compile.log
vlt/t_xml_first: Test PASSED
==SUMMARY: Left NO-FORKER  Passed 786  Unsup 15  Skipped 35  Failed 73
======================================================================
vlt/t_xml_tag: ==================================================
vlt/t_xml_tag: Compile
        perl ../bin/verilator --prefix Vt_xml_tag --x-assign unique -cc -Mdir obj_dir/t_xml_tag -OD --debug-check --comp-limit-members 10 --xml-only --clk clk  -f input.vc t/t_xml_tag.v    > obj_dir/t_xml_tag/vlt_compile.log
vlt/t_xml_tag: Test PASSED
==SUMMARY: Left NO-FORKER  Passed 787  Unsup 15  Skipped 35  Failed 73

======================================================================
TESTS Passed 787  Unsup 15  Skipped 35  Failed 73  Time 34:18
        #vlt/t_a_first_cc: %Error: Exec of perl failed
                make && test_regress/t/t_a_first_cc.pl --vlt
        #vlt/t_assert_comp_bad: %Error: Mismatch in output from obj_dir/t_assert_comp_bad/Vt_assert_comp_bad
                make && test_regress/t/t_assert_comp_bad.pl --vlt
        #vlt/t_assert_cover: %Error: Exec of perl failed
                make && test_regress/t/t_assert_cover.pl --vlt
        #vlt/t_assert_cover_off: %Error: Exec of perl failed
                make && test_regress/t/t_assert_cover_off.pl --vlt
        #vlt/t_assert_synth_full: %Error: Mismatch in output from obj_dir/t_assert_synth_full/Vt_assert_synth_full
                make && test_regress/t/t_assert_synth_full.pl --vlt
        #vlt/t_assert_synth_parallel: %Error: Mismatch in output from obj_dir/t_assert_synth_parallel/Vt_assert_synth_parallel
                make && test_regress/t/t_assert_synth_parallel.pl --vlt
        #vlt/t_clk_vecgen1: %Error: Exec of perl failed
                make && test_regress/t/t_clk_vecgen1.pl --vlt
        #vlt/t_clk_vecgen2: %Error: Exec of perl failed
                make && test_regress/t/t_clk_vecgen2.pl --vlt
        #vlt/t_clk_vecgen3: %Error: Exec of perl failed
                make && test_regress/t/t_clk_vecgen3.pl --vlt
        #vlt/t_crazy_sel: %Error: Exec of perl failed
                make && test_regress/t/t_crazy_sel.pl --vlt
        #vlt/t_debug_fatalsrc_bt_bad: %Error: Mismatch in output from perl
                make && test_regress/t/t_debug_fatalsrc_bt_bad.pl --vlt
        #vlt/t_debug_sigsegv_bt_bad: %Error: Mismatch in output from perl
                make && test_regress/t/t_debug_sigsegv_bt_bad.pl --vlt
        #vlt/t_dpi_imp_gen: %Error: Exec of perl failed
                make && test_regress/t/t_dpi_imp_gen.pl --vlt
        #vlt/t_final: %Error: Exec of perl failed
                make && test_regress/t/t_final.pl --vlt
        #vlt/t_flag_csplit: %Error: Exec of perl failed
                make && test_regress/t/t_flag_csplit.pl --vlt
        #vlt/t_func_gen: %Error: Exec of perl failed
                make && test_regress/t/t_func_gen.pl --vlt
[... 105 lines cut ...]
                make && test_regress/t/t_unroll_genf.pl --vlt
        #vlt/t_var_overwidth_bad: %Error: Mismatch in output from obj_dir/t_var_overwidth_bad/Vt_var_overwidth_bad
                make && test_regress/t/t_var_overwidth_bad.pl --vlt
        #vlt/t_verilated_threaded: %Error: Exec of make failed
                make && test_regress/t/t_verilated_threaded.pl --vlt
        #vlt/t_vgen: %Error: Please install Bit::Vector
                make && test_regress/t/t_vgen.pl --vlt
        #vlt/t_vpi_var: %Error: Exec of perl failed
                make && test_regress/t/t_vpi_var.pl --vlt
TESTS Passed 787  Unsup 15  Skipped 35  Failed 73  Time 34:18
make: *** [Makefile:49: test] Error 10
```
The verilator has been successfully installed and compiled in the Spack. However, 73 failures occur during the self-test during the make test. The compilation prompts specific problems. I cannot tell whether the source code is incorrect. Can you help me?

### Information on your system
[root@centos8 test_regress]# spack debug report
* **Spack:** 0.15.4-1365-35a9626ae
* **Python:** 3.6.8
* **Platform:** linux-centos8-aarch64

### Additional information

- [ ] I have run `spack debug report` and reported the version of Spack/Python/Platform
- [ ] I have searched the issues of this repo and believe this is not a duplicate
- [ ] I have run the failing commands in debug mode and reported the output

Tom-python0121, 2021-02-09:
```console
[root@centos8 test_regress]# ulimit -c unlimited
[root@centos8 test_regress]#  echo "core.%e.%p"> /proc/sys/kernel/core_pattern
[root@centos8 test_regress]# make test
[root@centos8 test_regress]# ls
Makefile                       core.Vt_uniqueif_fai..2896242  core.verilator_bin..2819499  core.verilator_bin..2833187  core.verilator_bin..2863632  core.verilator_bin..2900336
Makefile_obj                   core.Vt_uniqueif_fai..2896866  core.verilator_bin..2823360  core.verilator_bin..2833194  core.verilator_bin..2864205  core.verilator_bin..2907850
core.Vt_assert_basic..2804682  core.Vt_uniqueif_fai..2897629  core.verilator_bin..2823514  core.verilator_bin..2833218  core.verilator_bin..2865034  core.verilator_bin_d..2799889
core.Vt_assert_comp_..2805063  core.Vt_uniqueif_fai..2898035  core.verilator_bin..2828055  core.verilator_bin..2833378  core.verilator_bin..2865277  core.verilator_bin_d..2815155
core.Vt_assert_synth..2805862  core.Vt_unopt_conver..2899025  core.verilator_bin..2828337  core.verilator_bin..2833643  core.verilator_bin..2865284  core.verilator_bin_d..2815626
core.Vt_assert_synth..2806356  core.Vt_unopt_conver..2899388  core.verilator_bin..2831789  core.verilator_bin..2834058  core.verilator_bin..2865291  driver.pl
core.Vt_clk_latch..2813672     core.Vt_var_overwidt..2903857  core.verilator_bin..2831796  core.verilator_bin..2834065  core.verilator_bin..2865311  input.vc
core.Vt_initial_edge..2835798  core.verilator_bin..2805331    core.verilator_bin..2831800  core.verilator_bin..2839705  core.verilator_bin..2865339  obj_dir
core.Vt_order_clkins..2866321  core.verilator_bin..2805342    core.verilator_bin..2832584  core.verilator_bin..2840312  core.verilator_bin..2865362  t
core.Vt_order_double..2866629  core.verilator_bin..2813924    core.verilator_bin..2832588  core.verilator_bin..2840728  core.verilator_bin..2873377  vgen.pl
core.Vt_sys_readmem_..2886036  core.verilator_bin..2813931    core.verilator_bin..2832760  core.verilator_bin..2848406  core.verilator_bin..2873414
core.Vt_sys_readmem_..2886858  core.verilator_bin..2813938    core.verilator_bin..2833057  core.verilator_bin..2848414  core.verilator_bin..2880858
core.Vt_sys_readmem_..2887547  core.verilator_bin..2815079    core.verilator_bin..2833064  core.verilator_bin..2856680  core.verilator_bin..2898404
core.Vt_sys_readmem_..2888129  core.verilator_bin..2815598    core.verilator_bin..2833180  core.verilator_bin..2856701  core.verilator_bin..2898411
[root@centos8 test_regress]# cd ..
[root@centos8 spack-src]# find . -name "verilator_bin"
./bin/verilator_bin、
[root@centos8 bin]# ls
verilator  verilator_bin  verilator_bin_dbg  verilator_coverage  verilator_coverage_bin_dbg  verilator_difftree  verilator_includer  verilator_profcfunc
[root@centos8 test_regress]# ls
Makefile  Makefile_obj  core.verilator_bin_d.2775707  driver.pl  input.vc  obj_dir  t  vgen.pl

[root@centos8 test_regress]# gdb -c core.verilator_bin_d.2775707 ../bin/verilator_bin
NO
[root@centos8 test_regress]# gdb -c core.verilator_bin_d.2775707 ../bin/verilator_bin_dbg
GNU gdb (GDB) Red Hat Enterprise Linux 8.2-11.el8
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ../bin/verilator_bin_dbg...done.

warning: core file may not match specified executable file.
[New LWP 2775707]
Core was generated by `/tmp/root/spack-stage/spack-stage-verilator-3.920-74zk5x4hvr2bj2cwyb2btym32hy3g'.
Program terminated with signal SIGABRT, Aborted.
#0  0x0000ffffa0352c3c in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: yum debuginfo-install glibc-2.28-101.el8.aarch64 libgcc-8.3.1-5.el8.0.2.aarch64 libstdc++-8.3.1-5.el8.0.2.aarch64
(gdb) bt
#0  0x0000ffffa0352c3c in raise () from /lib64/libc.so.6
#1  0x0000ffffa03407a8 in abort () from /lib64/libc.so.6
#2  0x0000ffffa065a978 in std::__replacement_assert(char const*, int, char const*, char const*) () from /lib64/libstdc++.so.6
#3  0x0000ffffa06d4318 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator[](unsigned long) () from /lib64/libstdc++.so.6
#4  0x00000000005f4ce8 in VHashSha1::digestBinary[abi:cxx11]() (this=0xffffdcb7b0f8) at ../V3String.cpp:208
#5  0x00000000005f4dd8 in VHashSha1::digestHex[abi:cxx11]() (this=0xffffdcb7b0f8) at ../V3String.cpp:224
#6  0x00000000005f51cc in VHashSha1::selfTestOne (data="", data2="", exp="da39a3ee5e6b4b0d3255bfef95601890afd80709", exp64="2jmj7l5rSw0yVbBvlWAYkKBYBwk") at ../V3String.cpp:267
#7  0x00000000005f551c in VHashSha1::selfTest () at ../V3String.cpp:280
#8  0x0000000000407568 in main (argc=23, argv=0xffffdcb7bc38, env=0xffffdcb7bcf8) at ../Verilator.cpp:588
```
A large number of such source code problems have occurred. How can we solve these problems?

wsnyder, 2021-02-09:
First 3.920 is three years out of date and won't be fixed if there's something wrong (which there might not be), and second I don't use spack if you're going to use that you need to debug it or find someone that knows it, sorry.

I recommend using the most recent release and building as described in the installation instructions. https://github.com/verilator/verilator/blob/master/docs/install.adoc

wsnyder, 2021-02-09:
Closing as answered but please feel free to update if you get it working or find the issue.

closed 2021-02-09 by wsnyder