verilator/verilator #2785: verilator generate the VL_SHIFTL_WWQ function

qrq992, 2021-02-10:
hello,
verilator generate the VL_SHIFTL_WWQ function, But the function implementation does not exist.

demo：
module  test #(
    parameter EOP_WIDTH = 8,
    parameter DATA_WIDTH = 1024
)
(clk1, head_part0_dly4);
input clk1;
output [DATA_WIDTH-1:0]  head_part0_dly4;
localparam DATA_BYTE = DATA_WIDTH/8;
reg [DATA_WIDTH-1:0]  head_part0_dly4;
wire [DATA_WIDTH-1:0] mpi_head_use_dly4;
wire [EOP_WIDTH-2:0]   sop2_byte_dly4;

always @(posedge clk1) begin
    head_part0_dly4 <= mpi_head_use_dly4 << {(DATA_BYTE - sop2_byte_dly4), 3'd0};
end

endmodule

error message :
Vtest.cpp :  error: VL_SHIFTL_WWQ was not declared in this scope
    VL_SHIFTL_WWQ(1024,1024,35,__Vtemp1, test__DOOT__mpi_head_use_dly4, 

I read the source code ,  the code ' {(DATA_BYTE - sop2_byte_dly4), 3'd0}' should be generate function WWI, but the ‘3'd0’’(left shift)  make IData as QData.

wsnyder, 2021-02-11:
Adding the following to include/verilated.h should fit it.  I hit another problem trying to test this that I want to fix before committing it.

```
+static inline WDataOutP VL_SHIFTL_WWQ(int obits, int lbits, int rbits, WDataOutP owp, WDataInP lwp,
+                                      QData rd) VL_MT_SAFE {
+    WData rwp[VL_WQ_WORDS_E];
+    VL_SET_WQ(rwp, rd);
+    return VL_SHIFTL_WWW(obits, lbits, 32, owp, lwp, rwp);
+}

```

closed 2021-02-14 by wsnyder

qrq992, 2021-02-18:
It worked. Thanks.