verilator/verilator #2786: Consider enabling the GitHub discussions feature for this repo

mortbopet, 2021-02-10:
Currently, https://www.veripool.org/wiki/verilator referes to the Github issues page for forum activity.
I suggest enabling the (rather recent) GitHub Discussions feature. This provides a forum-like format where all discussions can take place, and by this, keep the 'issues' tab solely for actual bugs and feature requests.
If you find it relevant, the feature can be enabled in the repository settings.

wsnyder, 2021-02-13:
FWIW before moving to github Verilator had forums. One of the problems was maybe half of the forum issues ended up being bugs and there wasn't a way to convert to issues.  I played with the github discussions beta for a bit, and looks like this is planned but until [at least] that conversion feature is available I'd prefer to hold off.  Leaving open as a tickler for future consideration.

wsnyder, 2021-08-24:
At this time I think we'll stick with issues, as many discussions morph into issues.

closed 2021-08-24 by wsnyder