verilator/verilator #2788: Error in Interface arrays

mmatzev, 2021-02-14:
Hello, 

I use several arrays of interfaces, which should be handled in a always_comb statement. The interface itself is a fifo input/output as defined by  https://github.com/pulp-platform/hwpe-stream/tree/master/rtl

I use the following code, which should be easly unrollable:
`always_comb
begin : OBI_A_STREAM
    // HANDSHAKE
    // request if it is needed and the fifo is not full
    obi_host_h2d.req = ((req_buffer_in.receiver == IDLE) || ( fifo_input.ready == 1'b0)) ? 1'b0 : 1'b1; 
    // save it if the request is granted
    fifo_input.valid = obi_host_d2h.gnt;
    // give feedback to the input fifo
    for( integer i = 0; i < NB_CLIENTS; i++) begin
        cfg_addr_stream[i].ready  = ((req_buffer_in.receiver == CFG) && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
        addr_stream[i].ready      = ((req_buffer_in.receiver == RD)  && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
        wr_stream[i].ready        = ((req_buffer_in.receiver == WR)  && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
    end
...
`
NB_CLIENTS is a fix parameter.

The error message from verilator 4.108 is:

`%Error: hwac_manager_obi.sv:137:25: Expecting expression to be constant, but variable isn't const: 'i'
                                  : ... In instance hwac_top.i_hwac_manager.i_hwac_manager_obi
  137 |         cfg_addr_stream[i].ready  = ((req_buffer_in.receiver == CFG) && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
      |                         ^
%Error: hwac_manager_obi.sv:137:24: Could not expand constant selection inside dotted reference: 'i'
                                  : ... In instance hwac_top.i_hwac_manager.i_hwac_manager_obi
  137 |         cfg_addr_stream[i].ready  = ((req_buffer_in.receiver == CFG) && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
      |                        ^
%Error: hwac_manager_obi.sv:138:21: Expecting expression to be constant, but variable isn't const: 'i'
                                  : ... In instance hwac_top.i_hwac_manager.i_hwac_manager_obi
  138 |         addr_stream[i].ready      = ((req_buffer_in.receiver == RD)  && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
      |                     ^
%Error: hwac_manager_obi.sv:138:20: Could not expand constant selection inside dotted reference: 'i'
                                  : ... In instance hwac_top.i_hwac_manager.i_hwac_manager_obi
  138 |         addr_stream[i].ready      = ((req_buffer_in.receiver == RD)  && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
      |                    ^`

I had the same already in another place of the same module. My current work-around is to pass it not directly to the interface but to a signal array and then to use a generate statement to pass the signal array to the interface array.

Conclusion: Assignment and Access (the same behavior) works with generate and genvar but not with for loop inside of an always statement.

mmatzev, 2021-02-14:
Sorry, the code was not readable:
SystemVerilog:
always_comb
begin : OBI_A_STREAM
    // HANDSHAKE
    // request if it is needed and the fifo is not full
    obi_host_h2d.req = ((req_buffer_in.receiver == IDLE) || ( fifo_input.ready == 1'b0)) ? 1'b0 : 1'b1; 
    // save it if the request is granted
    fifo_input.valid = obi_host_d2h.gnt;
    // give feedback to the input fifo
    for( integer i = 0; i < NB_CLIENTS; i++) begin
        cfg_addr_stream[i].ready  = ((req_buffer_in.receiver == CFG) && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
        addr_stream[i].ready      = ((req_buffer_in.receiver == RD)  && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
        wr_stream[i].ready        = ((req_buffer_in.receiver == WR)  && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
    end

Verilator output:
%Error: hwac_manager_obi.sv:137:25: Expecting expression to be constant, but variable isn't const: 'i'
                                  : ... In instance hwac_top.i_hwac_manager.i_hwac_manager_obi
  137 |         cfg_addr_stream[i].ready  = ((req_buffer_in.receiver == CFG) && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
      |                         ^
%Error: hwac_manager_obi.sv:137:24: Could not expand constant selection inside dotted reference: 'i'
                                  : ... In instance hwac_top.i_hwac_manager.i_hwac_manager_obi
  137 |         cfg_addr_stream[i].ready  = ((req_buffer_in.receiver == CFG) && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
      |                        ^
%Error: hwac_manager_obi.sv:138:21: Expecting expression to be constant, but variable isn't const: 'i'
                                  : ... In instance hwac_top.i_hwac_manager.i_hwac_manager_obi
  138 |         addr_stream[i].ready      = ((req_buffer_in.receiver == RD)  && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
      |                     ^
%Error: hwac_manager_obi.sv:138:20: Could not expand constant selection inside dotted reference: 'i'
                                  : ... In instance hwac_top.i_hwac_manager.i_hwac_manager_obi
  138 |         addr_stream[i].ready      = ((req_buffer_in.receiver == RD)  && (req_buffer_in.client_id == i)) ? obi_host_d2h.gnt : 1'b0;
      |                    ^

wsnyder, 2021-02-14:
Verilator flattens all cell/interface arrays so a reference like cell[i] isn't supported. This means the "i" must be unrolled at generate time - which means "i" must be a genvar and in a generate loop.  If you can refactor the code to use a generate for loop it should work.

Fixing this is blocked by #1540.

wsnyder, 2024-09-14:
Duplicates #2116

closed 2024-09-14 by wsnyder