verilator/verilator #2793: Is it possible to get a user coverage count per instance instead of per module?

ekiwi, 2021-02-17:
My simple example circuit looks like this:
```v
module test(
  input        clock,
  input  [7:0] a,
  input  [7:0] b
);
  wire  c1_clock;
  wire [7:0] c1_a;
  wire  x2_clock;
  wire [7:0] x2_a;
  wire [8:0] _GEN_0 = a + b;
  child c1 (
    .clock(c1_clock),
    .a(c1_a)
  );
  child x2 (
    .clock(x2_clock),
    .a(x2_a)
  );
  assign c1_clock = clock;
  assign c1_a = a;
  assign x2_clock = clock;
  assign x2_a = b;
endmodule
module child(
  input        clock,
  input  [7:0] a
);
  TestPropery: cover property (@(posedge clock) a == 8'h0);
endmodule
```

I call verilator like this: `verilator -Wno-fatal --cc test.sv --coverage-user --trace --trace-coverage --exe top.cpp`

After running my test I get the following output in `coverage.dat`:
```
# SystemC::Coverage-3
C 'ftest.svl28n0pagev_user/childoTestProperyhTOP.test.*.TestPropery' 25
```

It seems like verilator combined the coverage points in `test.c1` and `test.c2`. I was wondering if instead it would be possible to get separate counts for `test.c1.TestProperty` and `test.c2.TestProperty`.

wsnyder, 2021-02-18:
The data is all in place to do per_instance counting as this was a SystemC coverage feature, If you want to experiment, you can just change this:

```
diff --git a/include/verilated_cov.cpp b/include/verilated_cov.cpp
index f351c3140..994721e25 100644
--- a/include/verilated_cov.cpp
+++ b/include/verilated_cov.cpp
@@ -356,7 +356,7 @@ public:
         for (const auto& itemp : m_items) {
             std::string name;
             std::string hier;
-            bool per_instance = false;
+            bool per_instance = true;

             for (int i = 0; i < MAX_KEYS; ++i) {
                 if (itemp->m_keys[i] != KEY_UNDEF) {
```

If you want finer control, there is no way at present to specify the per_instance attribute in Verilator that passes this through to creation of the cover point, as Verilator doesn't yet support the covergroup option.per_instance syntax.  I suppose we could add a /*verilator*/ meta tag in the meantime.  Do you need finer control?

ekiwi, 2021-02-18:
Thanks a lot for the quick response, Wilson!

>  Do you need finer control?

I do not need finer control since I am going to post process the coverage counts anyways. So as long as everything is per_instance that is exactly what I need.

Would it be possible to make `per_instance` an argument to the `write` method (with the default set to `false` to maintain old behavior)?
Ideally I would like to be able to enable `per_instance` coverage without modifying the current verilator installation.

ekiwi, 2021-02-18:
After reading a bit more of the source code: Is there any way for the C++ testbench to set the `VL_CIK_PER_INSTANCE` key of all coverage points to `1`?

An ideal solution to getting per_instance coverage would work with older versions of verilator and without modifying the local installation. If I could somehow get the test bench to add this key, it seems like that would fit the bill.

wsnyder, 2021-02-19:
In the Verilator generated code there's this:

    VL_COVER_INSERT(count32p,  "filename",filenamep,  "lineno",lineno,  "column",column,
        "hier",std::string(name())+hierp,  "page",pagep,  "comment",commentp,  (linescovp[0] ? "linescov" : ""), linescovp);

Which would instead need to be

    VL_COVER_INSERT(count32p,  "filename",filenamep,  "lineno",lineno,  "column",column,
        "hier",std::string(name())+hierp,  "page",pagep,  "comment",commentp,  (linescovp[0] ? "linescov" : ""), linescovp,
        "per_instance", 1);

There's no way I can think of to get this without changing Verilator or the include files.

ekiwi, 2021-02-19:
I think I will try to change the `VL_COVER_INSERT` since that would have me only modify the generated files and it will work on older verilator versions.

Looking forward, I would still be interested in a configuration option, either through a verilator flag, a C++ pre-processor define or a method/parameter accessible from the C++ testbench to turn on global per_instance coverage. Do you think something like that would be feasible?

wsnyder, 2021-02-19:
Any of verilator flags, verilator pragma, IEEE option. or C++ define could do it, it would just plumb that flag down into the generated cover insert statements.  IEEE would be nice but a lot of work, so I think a verilator pragma would be the best effort-reward.  Perhaps you'd be willing to contribute a patch to do this?  I can provide more details.

ekiwi, 2021-02-19:
> Perhaps you'd be willing to contribute a patch to do this? I can provide more details.

I could look into this, however, I need you to tell me which approach I should take and which files I would have to touch.

wsnyder, 2021-02-19:
- [ ] t_assert_cover.v can test this, e.g. add to Sub

   // Simple cover, per-instance
   // verilator coverage_per_instance
   pi_sub:
     cover property (@(posedge clk) cyc == 3);
   // Simple cover, not per-instance
   // verilator coverage_not_per_instance
   npi_sub:
     cover property (@(posedge clk) cyc == 3);

- [ ] Have t_assert_cover.pl check the new counts are correct, namely pi_sub above has per-instance counts, and npi_sub has combined counts
- [ ] document new pragmas in bin/verilator
- [ ] add coveragePerInstance to V3Error similar to coverageOn
- [ ] add pragma to src/verilog.l similar to verilator coverage_on/off
- [ ] in src/V3EmitC.cpp where output CoverDecl also output if perInstance is true/false
- [ ] in src/V3EmitC.cpp where __vlCoverInsert is declared take in the new perInstance flag and use it for the perInstance flag into coverage.
- [ ] Should also add VLT file support to tests, verilog.l, verilog.y similar to code that implements yVLT_COVERAGE_ON).

ekiwi, 2021-02-19:
Thanks! It will take me some time to work on this, but I will let you know once I make some progress.

ekiwi, 2021-03-19:
So I managed to solve this issue for my use-case.

I noticed that immediate cover statements are not merged accross the hierarchy.

So changing:
```v
  TestPropery: cover property (@(posedge clock) a == 8'h0);
```
to:
```v
  always @(posedge clock) begin
    cover(a == 8'h0);
  end
```

did the trick for me.

Now I get a `coverage.dat` like this:
```
# SystemC::Coverage-3
C 'ftest.svl31n0pagev_user/childocoverhTOP.test.c1' 0
C 'ftest.svl31n0pagev_user/childocoverhTOP.test.x2' 25
```

In case you are interested in my use-case: I am trying to support simulator independent coverage reports for our Chisel3 testing infrastructure: https://github.com/ucb-bar/chisel-testers2/pull/274

ekiwi, 2021-03-19:
Thanks for all the support!

closed 2021-03-19 by ekiwi

wsnyder, 2021-03-20:
I don't think its intentional that they behave differently, but hey, if it works.

ekiwi, 2021-03-20:
> I don't think its intentional that they behave differently, but hey, if it works.

I will be back once you change the behavior :-)

ekiwi, 2021-03-20:
Seems like this behavior also changed at some point between `4.028` (default on Ubuntu 20.04) and `4.032` (default on Fedora 33).

On the older version I get:
```
# SystemC::Coverage-3
C 'fTest1Module.svl40n0pagev_user/Test1ModuleocoverhTOP.Test1Module' 3
C 'fTest1Module.svl8n0pagev_user/SubModule1ocoverhTOP.Test1Module.c*' 0
```

and on the newer one:
```
# SystemC::Coverage-3
C 'fTest1Module.svl40n0pagev_user/Test1ModuleocoverhTOP.Test1Module' 3
C 'fTest1Module.svl8n0pagev_user/SubModule1ocoverhTOP.Test1Module.c0' 0
C 'fTest1Module.svl8n0pagev_user/SubModule1ocoverhTOP.Test1Module.c1' 0
```

I will continue investigating

ekiwi, 2021-03-20:
OK. Looks like I messed this up myself. I changed my local copy of `verilator`, forgot about it, and now that I am actually running in CI it comes back to bite me :)

wsnyder, 2021-03-20:
Can you easily call a Verilated function?

wsnyder, 2021-03-20:
in C++?

ekiwi, 2021-03-20:
> Can you easily call a Verilated function in C++?

I do control the C++ toplevel (we auto generate it) if that is what you mean.

wsnyder, 2021-03-20:
In master you can now call Verilated::defaultContextp()->forcePerInstance(true); just before writing coverage code.  See test_regress/t/t_cover_lib_c.cpp for testcode.  Example difference:

```
$ diff  test_regress/t/t_cover_lib_1{,_per_instance}.out
< C 'f../../t/t_cover_lib_c.cppl37pagesp_user/t_cover_lib_cP0htop.a*.pi' 500
---
> C 'f../../t/t_cover_lib_c.cppl37pagesp_user/t_cover_lib_cP0htop.a0.pi' 200
> C 'f../../t/t_cover_lib_c.cppl37pagesp_user/t_cover_lib_cP0htop.a1.pi' 300
```