verilator/verilator #2796: Is it possible to generate an HTML report considering toggle coverage?

Marc43, 2021-02-19:
How may we help - what is your question?
Hi,

I enabled `--coverage` option in my Verilator design. I have been able to have a `coverage.dat`, transform it with `verilator_coverage` and then create an HTML report. But this one only contains line coverage.

Any ideas?

Thanks!!
Marc

wsnyder, 2021-02-21:
When you look at the coverage.dat does it have lines with v_toggle, v_line, v_branch and v_user (if you have user statements)?  If so it is doing more than line coverage.

Marc43, 2021-02-22:
Thanks for answering. 

Sure, there are lines containing  v_toggle, v_line, v_branch. But, do I have to open the report and look at it through an editor or there's a way to include it in lcov for example or another visualizer tool?

How do you use it?

wsnyder, 2021-02-22:
If there are v_toggle etc then the coverage for those should be shown on the lines of the report that involve that coverage.  E.g. toggle coverage will be shown as the coverage percentage on the line declaring the signal, see e.g. lines 9 and 10 here https://codecov.io/gh/verilator/example-systemverilog/src/master/top.v

Marc43, 2021-02-22:
I can't access what you shared.

Nevertheless I checked the lcov report I have and indeed has numbers between line number and signal definition.

I can deduce this is toggle coverage:
[image]

But for v_branch? Is displayed in the same way? For example here:

[image]

wsnyder, 2021-02-22:
Yes, branch and user coverage are same way.

wsnyder, 2021-02-22:
Closing as assuming you have your answer, feel free to followup though if needed.

closed 2021-02-22 by wsnyder

Marc43, 2021-02-22:
Okay, that was useful, difficult to find documentation for things. Sorry if that was all newbie questions :) 

Thanks again!!!

wsnyder, 2021-02-22:
Great. There's a small paragraph in the verilator_coverage documentation.  Would certainly appreciate a pull request if you want to add more documentation details there or elsewhere.

### --write-info I<filename.info>

Specifies the aggregate coverage results, summed across all the files, should be written to the given filename in C<lcov> .info format. This may be used to use C<lcov> to aggregate or generate reports.

The info format loses data compared to the Verilator coverage data format; the info will all forms of coverage converted to line style coverage, and if there are multiple coverage points on a single line, the minimum coverage across those points will be used to report coverage of the line.