verilator/verilator #2798: Have seperate/no UNUSED warning for package parameters

GregAC, 2021-02-19:
v4.108 introduced UNUSED warnings for parameters (#2627). This unused warning is produced for an unused parameter no matter where it came from. In particular parameters that are imported from a package get an unused warning. A package may contain parameters a particular module or set of RTL doesn't use but are needed elsewhere so you can end up with many UNUSED warnings to waive.

UNUSED does seem more generally useful for module parameters and local parameters where lack of use is more likely to indicate a problem.

I suggest a separate UNUSED warning is created for parameters that come from a package or they don't generate UNUSED warnings at all.

I would be happy to do a PR for this. It appears V3Undriven.cpp is the place to do the change but I haven't managed to work out how to tell if the `AstVar` instance representing the parameter came from an imported package or elsewhere.

wsnyder, 2021-02-19:
I believe the current implementation will only warn of unused parameter packages if that parameter is unused across the entire package and all usages of that package, that is it has nothing to do with importing or not (it is not reported related to what modules use it, but rather as a whole).  If you don't want UNUSED inside packages, could you just turn off the lint message around the package's parameter declarations?

GregAC, 2021-02-22:
The specific scenario I'm thinking of is where you are building a simulation of one part of a system and importing some package you need some parameters from but not everything as it contains things that don't related to the part of the system you're simulating. You might also have auto-generated package contents where the generator produces a wide set of parameters for various usage scenarios but you only need a subset in the particular simulation you are building.

A concrete example is how we use Verilator to simulate OTBN (an in development accelerator for asymmetric cypto https://github.com/lowRISC/opentitan/tree/master/hw/ip/otbn) in OpenTitan (an SoC with many blocks of which OTBN is one). We have a Verilator testbench that simulates OTBN in isolation but the builds pulls in various OpenTitan package files, a mixture of hand written and auto-generated, that contain parameters relevant to a full OpenTitan system simulation but only some of them are used in the OTBN stand-alone simulation.

As an analogy to software imagine a software lint that warned on unused variables, would it be useful for it to report unused constants from library header files in the same way it reports other unused variables?

We could use the Verilator specific comments to turn off the UNUSED warnings around the package parameters though thus far have avoided such tool specific comments. There are various possibilities with the `lint_off` waiver rules and wildcards too. A simple, foolproof way to turn off unused warning on package parameters would be useful.

wsnyder, 2021-02-22:
Thanks, having a use case makes it easier to suggest good options.

Given what you are suggesting picking on just UNUSED parameters in packages doesn't seem quite right, there could easily be other present/future warnings in a package you don't want to see when "using" the package, and I don't think multiplying every one of those warnings by in-package/not-in-package is clean.

It still feels the best option is for the package to properly disable warnings (as I subscribe to the philosophy that files should always manage themselves), but can appreciate you might not want to do this.  BTW to suppress unused warnings in a tool-agnostic way I would recommend just roll them up to make them used somehow, e.g. like this:

             // Prevent unused warnings
             if (&{1'b0, PARAM1, PARAM2, ...}) begin end

Alternatively perhaps there should be a lint-off-in-packages flag, so you'd use that when editing user code that calls a package?

rswarbrick, 2021-02-22:
I wonder whether the right approach is to allow waivers with wildcards to catch this sort of thing? So the lint rules for the block that's partially using some packages would have entries like "ignore UNUSED warnings in my_other_pkg.sv". If we get that right, it won't be UNUSED-specific, avoiding the problem of proliferating "in-pkg" and "not-in-pkg" options.

> Alternatively perhaps there should be a lint-off-in-packages flag, so you'd use that when editing user code that calls a package?

I'm not a great fan of that. What if there's a silly width mismatch in the package code? We probably don't want to hide that sort of thing, since it's a bug in the package (that should be fixed!), rather than a property of which bits of the package happen to be used in this lint run.

wsnyder, 2021-02-22:
You can wildcard now, make a VLT file with:

   lint_off -rule UNUSED -file "my_other_pkg.sv"

GregAC, 2021-02-25:
We'll go with using the existing functionality available to deal with this so closing, thanks for the discussion.

closed 2021-02-25 by GregAC