verilator/verilator #2799: verilator@4.034：make test the test case fails to be built in.

Tom-python0121, 2021-02-20:
Hello,I meet a problem about make test the test case fails to be built in verilator@4.034
```console
[root@centos8 test_regress]# make test
……
-Skip: vltmt/t_xml_tag: scenario 'vltmt' not enabled for test
vltmt/t_xml_tag: %Skip: Skip: scenario 'vltmt' not enabled for test
==SUMMARY: Passed 1972  Failed 31  Skipped 145  Unsup 29  Time 142:50
==SUMMARY: Passed 1972  Failed 31  Skipped 145  Unsup 29  Time 142:50

======================================================================
        #dist/t_a4_examples: %Error: Exec of make -C ../examples/make_protect_lib failed:
                make && test_regress/t/t_a4_examples.pl  --dist
        #vlt/t_flag_wpedantic_bad: %Error: Line 1 mismatches; obj_vlt/t_flag_wpedantic_bad/vlt_compile.log != t/t_flag_wpedantic_bad.out
                make && test_regress/t/t_flag_wpedantic_bad.pl  --vlt
        #vlt/t_iff: %Error: Line 1 mismatches; obj_vlt/t_iff/vlt_compile.log != t/t_iff.out
                make && test_regress/t/t_iff.pl  --vlt
        #vlt/t_lint_bsspace_bad: %Error: Line 5 mismatches; obj_vlt/t_lint_bsspace_bad/vlt_compile.log != t/t_lint_bsspace_bad.out
                make && test_regress/t/t_lint_bsspace_bad.pl  --vlt
        #vlt/t_lint_comb_bad: %Error: Line 1 mismatches; obj_vlt/t_lint_comb_bad/vlt_compile.log != t/t_lint_comb_bad.out
                make && test_regress/t/t_lint_comb_bad.pl  --vlt
        #vlt/t_lint_in_inc_bad: %Error: Line 1 mismatches; obj_vlt/t_lint_in_inc_bad/vlt_compile.log != t/t_lint_in_inc_bad.out
                make && test_regress/t/t_lint_in_inc_bad.pl  --vlt
        #vlt/t_lint_mod_paren_bad: %Error: Line 1 mismatches; obj_vlt/t_lint_mod_paren_bad/vlt_compile.log != t/t_lint_mod_paren_bad.out
                make && test_regress/t/t_lint_mod_paren_bad.pl  --vlt
        #vlt/t_lint_pkg_colon_bad: %Error: Line 1 mismatches; obj_vlt/t_lint_pkg_colon_bad/vlt_compile.log != t/t_lint_pkg_colon_bad.out
                make && test_regress/t/t_lint_pkg_colon_bad.pl  --vlt
        #vlt/t_lint_rsvd_bad: %Error: Line 4 mismatches; obj_vlt/t_lint_rsvd_bad/vlt_compile.log != t/t_lint_rsvd_bad.out
                make && test_regress/t/t_lint_rsvd_bad.pl  --vlt
        #vlt/t_param_type_bad2: %Error: Line 1 mismatches; obj_vlt/t_param_type_bad2/vlt_compile.log != t/t_param_type_bad2.out
                make && test_regress/t/t_param_type_bad2.pl  --vlt
        #vlt/t_param_type_bad: %Error: Line 1 mismatches; obj_vlt/t_param_type_bad/vlt_compile.log != t/t_param_type_bad.out
                make && test_regress/t/t_param_type_bad.pl  --vlt
        #vlt/t_pp_circ_subst_bad: %Error: Line 2 mismatches; obj_vlt/t_pp_circ_subst_bad/vlt_compile.log != t/t_pp_circ_subst_bad.out
                make && test_regress/t/t_pp_circ_subst_bad.pl  --vlt
        #vlt/t_pp_circdef_bad: %Error: Line 2 mismatches; obj_vlt/t_pp_circdef_bad/vlt_compile.log != t/t_pp_circdef_bad.out
                make && test_regress/t/t_pp_circdef_bad.pl  --vlt
        #vlt/t_pp_defparen_bad: %Error: Line 4 mismatches; obj_vlt/t_pp_defparen_bad/vlt_compile.log != t/t_pp_defparen_bad.out
                make && test_regress/t/t_pp_defparen_bad.pl  --vlt
        #vlt/t_pp_resetall_bad: %Error: Line 1 mismatches; obj_vlt/t_pp_resetall_bad/vlt_compile.log != t/t_pp_resetall_bad.out
                make && test_regress/t/t_pp_resetall_bad.pl  --vlt
        #vlt/t_preproc_inc_bad: %Error: Line 1 mismatches; obj_vlt/t_preproc_inc_bad/vlt_compile.log != t/t_preproc_inc_bad.out
                make && test_regress/t/t_preproc_inc_bad.pl  --vlt
        #vlt/t_prot_lib: %Error: Exec of make failed: /home/spack/spack/lib/spack/env/gcc/g++  -I.  -MMD -I/tmp/root/spack-stage/spack-stage-verilator-4.034-qfyeoadkxr65m4sn6owh7c6a3oxootby/spack-src/test_regress/../include -I/tmp/root/spack-stage/spack-stage-verilator-4.034-qfyeoadkxr65m4sn6owh7c6a3oxootby/spack-src/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_prot_lib -DVM_PREFIX=Vt_prot_lib -DVM_PREFIX_INCLUDE="<Vt_prot_lib.h>" -DT_PROT_LIB    -c -o Vt_prot_lib__main.o ../../obj_vlt/t_prot_lib/Vt_prot_lib__main.cpp
                make && test_regress/t/t_prot_lib.pl  --vlt
        #vlt/t_prot_lib_clk_gated: %Error: Exec of make failed: /home/spack/spack/lib/spack/env/gcc/g++  -I.  -MMD -I/tmp/root/spack-stage/spack-stage-verilator-4.034-qfyeoadkxr65m4sn6owh7c6a3oxootby/spack-src/test_regress/../include -I/tmp/root/spack-stage/spack-stage-verilator-4.034-qfyeoadkxr65m4sn6owh7c6a3oxootby/spack-src/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_prot_lib_clk_gated -DVM_PREFIX=Vt_prot_lib_clk_gated -DVM_PREFIX_INCLUDE="<Vt_prot_lib_clk_gated.h>" -DT_PROT_LIB_CLK_GATED    -c -o Vt_prot_lib_clk_gated__main.o ../../obj_vlt/t_prot_lib_clk_gated/Vt_prot_lib_clk_gated__main.cpp
                make && test_regress/t/t_prot_lib_clk_gated.pl  --vlt
        #vlt/t_time_vpi_10ms10ns: %Error: Line 13 mismatches; obj_vlt/t_time_vpi_10ms10ns/vlt_sim.log != t/t_time_vpi_10ms10ns.out
                make && test_regress/t/t_time_vpi_10ms10ns.pl  --vlt
        #vlt/t_time_vpi_1fs1fs: %Error: Line 13 mismatches; obj_vlt/t_time_vpi_1fs1fs/vlt_sim.log != t/t_time_vpi_1fs1fs.out
                make && test_regress/t/t_time_vpi_1fs1fs.pl  --vlt
        #vlt/t_time_vpi_1ms10ns: %Error: Line 13 mismatches; obj_vlt/t_time_vpi_1ms10ns/vlt_sim.log != t/t_time_vpi_1ms10ns.out
                make && test_regress/t/t_time_vpi_1ms10ns.pl  --vlt
        #vlt/t_time_vpi_1ns1ns: %Error: Line 13 mismatches; obj_vlt/t_time_vpi_1ns1ns/vlt_sim.log != t/t_time_vpi_1ns1ns.out
                make && test_regress/t/t_time_vpi_1ns1ns.pl  --vlt
        #vlt/t_time_vpi_1ps1fs: %Error: Line 13 mismatches; obj_vlt/t_time_vpi_1ps1fs/vlt_sim.log != t/t_time_vpi_1ps1fs.out
                make && test_regress/t/t_time_vpi_1ps1fs.pl  --vlt
        #vlt/t_time_vpi_1us1ns: %Error: Line 13 mismatches; obj_vlt/t_time_vpi_1us1ns/vlt_sim.log != t/t_time_vpi_1us1ns.out
                make && test_regress/t/t_time_vpi_1us1ns.pl  --vlt
        #vlt/t_udp_noname: %Error: Line 1 mismatches; obj_vlt/t_udp_noname/vlt_compile.log != t/t_udp_noname.out
                make && test_regress/t/t_udp_noname.pl  --vlt
        #vlt/t_unconnected_bad: %Error: Line 7 mismatches; obj_vlt/t_unconnected_bad/vlt_compile.log != t/t_unconnected_bad.out
                make && test_regress/t/t_unconnected_bad.pl  --vlt
        #vlt/t_var_bad_sv: %Error: Line 8 mismatches; obj_vlt/t_var_bad_sv/vlt_compile.log != t/t_var_bad_sv.out
                make && test_regress/t/t_var_bad_sv.pl  --vlt
        #vltmt/t_gantt: %Error: Exec of obj_vltmt/t_gantt/Vt_gantt failed: %Warning: vltmt/t_gantt: Exec failed with core dump
                make && test_regress/t/t_gantt.pl  --vltmt
        #vltmt/t_iff: %Error: Line 1 mismatches; obj_vltmt/t_iff/vlt_compile.log != t/t_iff.out
                make && test_regress/t/t_iff.pl  --vltmt
        #vltmt/t_lint_in_inc_bad: %Error: Line 1 mismatches; obj_vltmt/t_lint_in_inc_bad/vlt_compile.log != t/t_lint_in_inc_bad.out
                make && test_regress/t/t_lint_in_inc_bad.pl  --vltmt
        #vltmt/t_udp_noname: %Error: Line 1 mismatches; obj_vltmt/t_udp_noname/vlt_compile.log != t/t_udp_noname.out
……
```
I know they are coredump problems. What are the solutions for this version?

wsnyder, 2021-02-20:
Why is it coredumping?

    test_regress/t/t_flag_wpedantic_bad.pl  --debug --gdbbt

wsnyder, 2021-02-20:
Also as you are on a very old version, please try using the git sources to see if that fixes your problem.

wsnyder, 2021-03-07:
I am assuming you figured this out, otherwise feel free to post more information and reopen.

closed 2021-03-07 by wsnyder